lopsided98/nix-ros-overlay #541: Some `moveit-ros-*` packages builds time out.

liarokapisv, 2024-12-13:
Due to the caching issues, I recently had to build many packages by source and I noticed that a few packages stand out. Specifically, packages from `moveit-ros`, like `moveit-ros-move-group` and `moveit-ros-sensors` for some reason take a ton of time to build, I pretty much gave up after 2 hours. It seems these also time-out in the Hydra instance which may be relevant to the CI issues as well. The issues appears to be at the end of the build phase, for example see the logs at https://hydra.iid.ciirc.cvut.cz/build/297469/nixlog/1

I am currently testing with:

```bash
Resolved URL:  git+file:///home/veritas/acumino_ws
Description:   MELFA ROS 2 Driver
Path:          /nix/store/xll3klidn3ca6lls6vzq3d37ngzyym72-source
Revision:      6155c3720068f74e0a578ef85460edd4a0109565-dirty
Last modified: 2024-12-12 13:17:17
Inputs:
├───nixpkgs follows input 'ros-overlay/nixpkgs'
├───nixpkgs-unstable: github:nixos/nixpkgs/a73246e2eef4c6ed172979932bc80e1404ba2d56?narHash=sha256-463SNPWmz46iLzJKRzO3Q2b0Aurff3U1n0nYItxq7jU%3D (2024-12-09 15:59:59)
├───ros-overlay: github:lopsided98/nix-ros-overlay/c3e6e964fcdd917193dca68cc5cb16f128fe5e1b?narHash=sha256-TzKPxouvrol38ng/VmHkNxObdon0taXHhcPmo/09cJo%3D (2024-11-16 03:33:31)
│   ├───flake-utils: github:numtide/flake-utils/c1dfcf08411b08f6b8615f7d8971a2bfa81d5e8a?narHash=sha256-X6rJYSESBVr3hBoH0WbKE5KvhPU5bloyZ2L4K60/fPQ%3D (2024-09-17 08:14:13)
│   │   └───systems: github:nix-systems/default/da67096a3b9bf56a91d16901293e51ba5b49a27e?narHash=sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768%3D (2023-04-09 08:27:08)
│   └───nixpkgs: github:lopsided98/nixpkgs/345c263f2f53a3710abe117f28a5cb86d0ba4059?narHash=sha256-sjiGsMh%2B1cWXb53Tecsm4skyFNag33GPbVgCdfj3n9I%3D (2024-09-13 16:03:24)
└───ros-workspace: github:hacker1024/nix-ros-workspace/d389c0ad68267b85b9ac3b7350801dd0110ed669?narHash=sha256-FXwNfiKBBarRjo4apAZuBU/OVgrNWa/lwi%2BS8HwHbvY%3D (2024-03-27 02:30:11)
```

lopsided98, 2024-12-13:
Why doesn't upstream run into this issue? Is there something Nix specific that exacerbates this problem?

liarokapisv, 2024-12-14:
I think the reason is that nix implicitely (using the `cc-wrapper` setup hook) passes all dependencies available to the `NIX_LDFLAGS`. This is often not an issue for `nixpkgs` specifically because it doesn't use `propagatedNativeBuildInputs` much so not that many dependencies end up at each level. Because `nix-ros-overlay` makes heavy usage of this, the `NIX_LDFLAGS` ends up becoming very large (eg 100/200+ libraries) so when we use `ld` to filter symbols `--as-needed` it ends up searching too many libraries.

closed 2025-01-01 by lopsided98