lopsided98/nix-ros-overlay #543: How are colcon extensions added to a flake?

Guelakais, 2024-12-13:
In fact, I'm a bit confused by the way packages are integrated by the nix-ros-overlay. Simple example:
`colcon` is entered normally in the pkgs list. With `ament-cmake-cargo` this no longer works. This must then be included in a list with `pkgs.rospackages.humble`. Unfortunately, I am now completely out of the Colcon extensions. Specifically: How are `colcon-cargo` and `colcon-ros-cargo` integrated? That's the flake I'm working on at the moment:
```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs-unstable.url = "github:NixOS/nixpkgs/nixos-unstable";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs"; # IMPORTANT!!!
  };
  outputs =
    {
      self,
      nix-ros-overlay,
      nixpkgs,
      nixpkgs-unstable,

    }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
        pkgs-unstable = import nixpkgs-unstable {
          inherit system;
        };
      in
      {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = with pkgs; [
            colcon
            (
              with pkgs.rosPackages.humble;
              buildEnv {
                paths = [
                  ros-core
                  rosidl-generator-rs
                  ament-cmake-core
                ];
              }
            )
            (with pkgs.python312Packages; [
              argcomplete
              polars
              numpy
              altair
              matplotlib
              pip

            ])
            (with pkgs-unstable; [
              rustc
              bash-completion
              curl
              wget
              git
              clang
              libclang
              vcstool
              uv
              cargo-outdated
            ])

          ];
        };
      }
    );
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

wentasah, 2024-12-13:
This is how I'm using `colcon-ros-cargo`:
```nix
  ...
  packages = [
    pkgs.bashInteractive
    (pkgs.colcon.withExtensions [ pkgs.python3Packages.colcon-ros-cargo ])
    pkgs.cargo-ament-build
    (rosPkgs.buildEnv {
       ...
```
Does it work for you?

Guelakais, 2024-12-13:
When you showed me this, I immediately suspected that the colcon extensions were simply packaged as normal python packages. In fact, they are:
```nix
(with pkgs.python312Packages; [
argcomplete
colcon-argcomplete
colcon-cargo
colcon-devtools
colcon-defaults
colcon-mixin
colcon-ros-cargo
polars
numpy
altair
matplotlib
pip
])
```
Conceptually, the same principle applies to nix as to rust: if it compiles, it works. Accordingly, you can do it that way and it works.