lopsided98/nix-ros-overlay #544: Adding all dependencies to `propagatedBuildInputs` is not sustainable.

liarokapisv, 2024-12-13:
The heavy usage of `propagatedBuildInputs` means that projects with many dependencies end up collecting a ton of gcc flags through the `cc-wrapper`.  

This includes - for each dependency in the dependency chain - the include folder path and the lib folder path together.
Each path is ~80 characters on average.

When using a dev-shell for a workspace, it will pickup a ton of transitive dependencies from all the workspace packages.

As an example, my devshell environment ends up with ~260 dependencies (because they are transitive).

For some reason these get a x3 duplication in the cc-wrapper  invocation, probably due to a cc-wrapper setup-hook.sh omition when `propagatedBuildInputs` or something, I am still investigating the reason.

This means that any gcc invocation ends up with x3 x260 x2 x80 = 124800 kb worth of flags. 

Add all the extra `-system` and additional gcc crafts and we end up hitting https://github.com/NixOS/nixpkgs/issues/41340

This was not an issue for me with an older `master` commit, I am guessing that some packages just added a few extra transitive dependencies and it managed to tip the balance.

liarokapisv, 2024-12-14:
It appears that the amplification happens due to `cc-wrapper` setup hooks being injected at different levels. Specifically the problem seems to appear due to `clang` being used for the various ament tools.

See relevant issue at https://github.com/NixOS/nixpkgs/issues/364984.

hacker1024, 2024-12-17:
To my understanding, we end up with many `propagatedBuildInputs` because it's often impossible to know if a dependency actually needs to be propagated or not. [REP 149](https://www.ros.org/reps/rep-0149.html#dependency-tags) actually defines a wide range of dependency types that can map pretty nicely to Nixpkgs's, but a significant amount of packages use the basic [`depend`](https://www.ros.org/reps/rep-0149.html#depend-multiple) tag, which is not helpful.

(See [here](https://github.com/lopsided98/superflore/blob/nixos-support/superflore/generators/nix/nix_package.py#L85-L111) for the current Nix generation logic.)

I wonder how necessary this actually is. What would happen if we tried to use `buildInputs` instead? If there are certain common dependencies that do need to be propagated, could we get away with maintaining a list of them and only propagating packages from that whitelist by default? At the very least, we can use a blacklist - I doubt `rclcpp` ever needs to be propagated, for example, but it often is.

I also wonder if we can handle the [`exec_depend`](https://www.ros.org/reps/rep-0149.html#exec-depend-multiple) group a little better:

> The <exec_depend> is needed for packages providing shared libraries, executable commands, Python modules, launch scripts or any other files required for running your package. It is also used by metapackages for grouping packages.

At the moment, we add `exec_depend` to `nativeBuildInputs`. If you think about it, though:
- Shared libraries will be linked to directly in the far majority of cases, so they're fine in `buildInputs` alone. We can default to this.
- Executable commands can be provided through wrappers adding to PATH instead of generic propagation. These packages can be identified by scanning their `bin` directories for executables.
- Python modules are a legitimate usecase, but we can identify them pretty easily and add them to `propagatedBuildInputs` (or the new `dependencies` attribute`) as normal.
- Launch scripts are harder, but they aren't used during building. We can add ROS packages to a different list in `passthru` instead that is then used by `buildEnv` directly.

This `exec_depend` thing is really the main point of friction between REP 149 and Nixpkgs. On a regular Linux system, shared libraries are runtime dependencies that must be installed alongside the intended package. With Nix, that happens automatically, so they're a `buildInput` and not a `propagatedBuildInput`. Executable dependencies are similar (included in wrappers, so a `buildInput` and not a `propagatedBuildInput`).

I wonder if we can convince the ROS package manifest format team to make some changes that would help us and improve the ecosystem.

- Split `exec_depend` into its four categories: `exec_lib_depend`, `exec_bin_depend`, `exec_python_depend`, `exec_ros_depend`
- Deprecate the generic `depend` so that package maintainers become more granular in their `package.xml`s

mikepurvis, 2026-09-24:
As a more significant departure from nix ecosystem norms, what I did about this at Clearpath / OTTO (~2000 pkg workspace) was:

- ignore propagatedBuildInputs
- don't wrap executables, so "naked" package installspaces aren't very useful on their own
- carry the exact semantic info from the package example in passthru attrs like buildDepends, runDepends
- assemble the runtime or devtime package closure _at workspace assembly time_
- optionally assembly a workspace-specific symlinkjoin of all those packages so that ROS_PACKAGE_PATH, PYTHONPATH, and friends all just carry a single entry rather than hundreds (critical esp for Python interpreter startup time).
- all in-nix builds get routed through colcon so that we end up with the colcon dsv path/hook files and can specify the exact build config we need consistently (eg arch optimizations, alignment, etc) between nix-delivered packages and the developer-build overlay workspaces.

Here's the relevant `buildColconPackage` section:

https://github.com/clearpathrobotics/nix-ros-base/blob/da411f63d3856fa40fd3716ce1976493bac7a72f/lib/build-colcon-package.nix#L186-L200

The workspace-realization part of this is here:

https://github.com/clearpathrobotics/nix-ros-base/blob/devel/lib/build-colcon-workspace.nix

As I say, I don't know how this approach could be reconciled with the normal conventions of how software is packaged for Nix, but it _is_ what worked for delivering unbloated run-time and dev-time workspaces with large numbers of packages. Would be happy to see any of it put to use.

liarokapisv, 2026-09-25:
This sounds like a much more scalable approach. The whole nixpkgs env injection is big hack to make single-package installs possible. 
What we need is indeed properly model the dependencies and reconcile at a the final closure creation. Nixpkgs also wants packages to be installable standalone in dev shells so there is no single place to fix the dependency tree.  Here we have the workspace abstraction which makes sense as a reconciliation point, I think it's the right approach.

wentasah, 2026-09-25:
There are two things that must be solved:

1. Building packages on their own, independently of others.
2. Using packages in development workspace (as underlay in colcon terminology)

Do I understand correctly that your approach is is primarily about the second?

> As a more significant departure from nix ecosystem norms, what I did about this at Clearpath / OTTO (~2000 pkg workspace) was:
> 
>     * ignore propagatedBuildInputs

Yes, this is the source of the problem. But if we disable them completely, we won't build anything, because dependencies from nixpkgs will be missing.
> 
>     * don't wrap executables, so "naked" package installspaces aren't very useful on their own

Currently, we don't wrap most of them. Only Qt apps are wrapped explicitly. Then we use `buildEnv`, which build "ROS workspace" and wraps them.

>     * carry the exact dep info from the package.xml in passthru attrs named for the ROS-side semantics:  `buildDepends`, `runDepends`, `testDepends`

This is probably the crucial missing point here. We use many things as `propagatedBuildInputs`, but likely, only `<build_export_depend>` should be used for this. `<exec_depend>` is only a workspace, i.e. `buildEnv` thing.

>     * assemble the runtime or devtime package closure _at workspace assembly time_

Does this mean that you built every package differently for each workspace it is included in? Like create devtime closure from the workspace and build all packages in the workspace against it?
 
>     * optionally assemble a workspace-specific symlinkjoin of all those packages so that ROS_PACKAGE_PATH, PYTHONPATH, and friends all just carry a single entry rather than hundreds (critical esp for Python interpreter startup time).

This we already do in `buildEnv`.

>     * all in-nix builds get routed through colcon so that we end up with the colcon dsv path/hook files and can specify the exact build config we need consistently (eg arch optimizations, alignment, etc) between nix-delivered packages and the developer-build overlay workspaces.

We route everything through `buildRosPackage`, which does not use colcon, but can do whatever we need (which is almost nothing today). I'm wondering about these colcon dsv files. Isn't it more or less the same thing as `nix-support/propagate-build-inputs` file?
 
> Here's the relevant `buildColconPackage` section:
> 
> https://github.com/clearpathrobotics/nix-ros-base/blob/da411f63d3856fa40fd3716ce1976493bac7a72f/lib/build-colcon-package.nix#L186-L200
> 
> The workspace-realization part of this is here:
> 
> https://github.com/clearpathrobotics/nix-ros-base/blob/devel/lib/build-colcon-workspace.nix
> 
> As I say, I don't know how this approach could be reconciled with the normal conventions of how software is packaged for Nix, but it _is_ what worked for delivering unbloated run-time and dev-time workspaces with large numbers of packages. Would be happy to see any of it put to use. At the very least, attaching the passthru attributes to the generated package definitions might give a foothold to begin experimenting with workspace-time dep propagation.

I'll look at it again. There might be some useful things. But no ETA.

Another thing I was thinking about was to reduce the number of propagated builds by building one `symlinkJoin` for all propagated dependencies so each package would propagate at most one path. This would probably reduce the length of environment variables and command switches at the cost more many more inodes allocated. Therefore reducing what goas into propagatedBuildInputs is likely better.

BTW Is anybody at NixCon to discuss these and other things?

mikepurvis, 2026-09-25:
> Does this mean that you built every package differently for each workspace it is included in? Like create devtime closure from the workspace and build all packages in the workspace against it?

Oh no, every package is built once, for all workspaces (package xml conditionals were handled at the dist level in my implementation, so those would obviously build twice, say if a package could be ROS 1 or ROS 2, but that's a separate matter).

But yeah, every package is built-in-Nix against "fat" envvars, so `CMAKE_PREFIX_PATH` and friends all pointing at the real store paths of the dependencies. This ensures that the executable library paths and so-on are pointing at a long-lived path and not a workspace. The symlinkjoin happens when a package and its deps are brought together either with recursive run deps or optionally with some pool of the build deps to create a usable environment for normal ROS developer multi-package incremental workspace builds.

Another nice side effect not noted above is that run deps are no longer required at build time, which at least in the OTTO tree meant significant gains in being able to parallelize parts of the build that invoked each other's binaries (= exec-depend) but didn't actually need to build depend.

I guess the closest analogue to this in nixpkgs might be ecosystems like Python. If I `nix build nixpkgs#python3Packages.pyyaml`, the result of that isn't really something that's much use on its own; it only exists as a building block to assemble a workspace (`python3.withPackages`) or an application (`python3Packages.buildPythonApplication`). It's not a huge stretch to think of ROS workspaces and apps in similar terms.

For the purposes of that demo, I actually attached a `ws` attribute to every package, so you could build `nix build github:clearpathrobotics/nix-ros#noetic.ros_base` and that would get you just the (empty) metapackage, but if you did `nix develop github:clearpathrobotics/nix-ros#noetic.ros_base.ws` _then_ you would have the ros_base workspace with all its deps (that demo is unfortunately heavily stale since it was a one-time code drop four years ago, but of course being Nix it's still perfectly buildable from source). That code drop unfortunately pre-dates it, but we had a later version that was like `dist.pkg.ws.setup-sh` that would just render out the setup script directly, to fully control the workspace without all the extra gunk that `nix develop` throws in on top.

> I'm wondering about these colcon dsv files. Isn't it more or less the same thing as `nix-support/propagate-build-inputs` file?

Kind of yeah, but they bind later; the paths in there are relative to the package's installspace rather than being absolute. That makes them friendly to packages being built in isolation and then either run that way or copied/symlinked into a merged workspace and the workspace-assembly just transparently seeing that now all the dsvs in the merged directory want the same python directory or whatever. I think this was mainly done for performance as it's obviously faster than per-package shell hooks that search the envvar before deciding to append or not, but it worked nicely for also letting my package installspaces serve double duty: they work in isolation when building another package for Nix, and they work merged when creating a dev or runtime workspace.

Re NixCon, sadly I won't be there, but happy to jump on a

mikepurvis, 2026-09-25:
Just to also quickly add, obviously the above scheme does help a lot with workspace performance, size, and build parallelism, but build deps still are carried implicitly via store paths baked into the CMake modules. For roscpp, for example, much of what is in the list is what you would expect, but there are still dependency dev packages present:

https://gist.github.com/mikepurvis/2e3a7a660a5c93b7f1c36e3fc97dd707

And this is how it happens:

```
$ nix build github:clearpathrobotics/nix-ros#noetic.roscpp
mikepurvis@meridian:~$ cat result/share/roscpp/cmake/* | grep include_dirs
  set(_include_dirs "include;/nix/store/xjrz31plbx0iz5k9yxq468w0khbfhbgb-boost-1.79.0-dev/include")     # noooo
  foreach(idir ${_include_dirs})
```

We actually did pursue a full dev/out/dbg split for every package at OTTO, and that absolutely worked too, though it required considerably more fixup hackery, particularly on the ROS 2 side. See https://github.com/lopsided98/nix-ros-overlay/issues/698#issuecomment-3406480351 and https://discourse.openrobotics.org/t/proposal-for-gnuinstalldirs-use-to-enable-multiple-installspaces/29182

At ROSCon this week @cottsay (current ros infra maintainer) said he was starting on separate dev RPMs for Fedora, so although that's a considerably easier problem to solve (since they come back together on disk unlike with Nix where they need to _build and run_ in their separated form), it may be an opportunity to make movements in the right direction if there's rethinking of these bits anyway.

mikepurvis, 2026-09-29:
Claude and I took a quick stab at this, here's the MVP of what it looks like against the current nix-ros-overlay codebase; the superflore side is fairly minimal, though obviously it's a compatibility and hash break with the existing generated packages:

https://github.com/mikepurvis/superflore/compare/aff8023...mikepurvis:superflore:nix-passthru-deps

On the consumption side, the basic build is also fairly, slim, but of course the change is massive because of the superflore run and I also smuggled in a setup-sh generator:

https://github.com/lopsided98/nix-ros-overlay/compare/develop...mikepurvis:nix-ros-overlay:passthru-exec-deps

So the quickie demo is:

```
$ nix build github:mikepurvis/nix-ros-overlay/passthru-exec-deps#jazzy.ros-base.ws.setup-sh

$ source result

$ which ros2
/nix/store/nc7slk3dbky1ly1cma4nfnd6mhba2n8a-ros-env/bin/ros2

$ ros2 topic list
/parameter_events
/rosout
```

In terms of the numbers, interestingly there is _some_ breaking up of build serialization, but at least within the core distro corpus, it's basically a wash:

> - In desktop_full the critical path is the Gazebo vendor chain (gz-math → gz-msgs → gz-transport → gz-gui → gz-sim, with gz-sim-vendor alone at 21 minutes). Every link is a real build dependency, so both layouts share it.
> - In desktop it's the fastrtps → rmw → rcl → rclcpp → rviz_common → rviz_default_plugins spine, also genuine build deps. The one minute saved is rviz2 no longer waiting on rviz-default-plugins.
> - The chain-length improvement (36 → 28 packages) is real but made of cheap packages, so it barely shows in minutes. In upstream ROS 2 the heavy builds are linked by true build deps, which is presumably different from OTTO's tree.

The more significant argument is around controlling rebuild fanout:

> - Median 10 → 4 (−60%); mean 70 → 60 (−15%); total rebuild work −14%.
> - Leaf-side packages stop rebuilding anything downstream: rviz-default-plugins 4 → 0, ros2launch 7 → 0, launch-ros 18 → 6.
> - The rosidl toolchain drops about 35 each: rosidl-cli 219 → 183, rosidl-parser 217 → 185.
> - Core packages drop 15–25%: rclcpp 98 → 80, rclpy 106 → 82.

And obviously, the deeper a stack is (eg for a commercial entity looking to deliver a whole ROS-based product on top of this), the more these savings pay off and potentially stack with eventual dev package separation.