lopsided98/nix-ros-overlay #545: rviz2: Cannot mix incompatible Qt library (5.15.15) with this library (5.15.14)

okvik, 2024-12-14:
```
$ rviz2
[INFO] [1734180025.817295354] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1734180025.817322395] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1734180025.837859214] [rviz2]: Stereo is NOT SUPPORTED
Cannot mix incompatible Qt library (5.15.15) with this library (5.15.14)
Aborted (core dumped)
```

This happens on both master and develop branches with Jazzy; I haven't tested other distros. Other Qt ROS stuff like `rqt` works fine for example.

I don't know what the root cause of the problem is, but it seems that something in my system configuration sets the Qt environment variables to 5.15.15, while rviz2 in the overlay expects 5.15.14.

I set my system configuration to follow nixpkgs from nix-ros-overlay, but it doesn't make a difference.

I managed to work around the issue by patching the environment variables in my development shell as follows:

```
export QT_PLUGIN_PATH=`echo $QT_PLUGIN_PATH | sed 's/qt-5.15.15/qt-5.15.14/g'`
```

movefasta, 2024-12-15:
Are you on NixOS?

okvik, 2024-12-15:
> Are you on NixOS?

Yes.

muellerbernd, 2025-09-15:
I have the same problem right now. This also causes problems while starting `gz sim`. @okvik your solution also works for me.
Maybe @wentasah can help here?

muellerbernd, 2025-09-15:
> How to trigger this problem? If I run `NIXPKGS_ALLOW_INSECURE=1 nix develop --impure github:lopsided98/nix-ros-overlay#example-ros2-gz` I can run `gz sim` and the opening screen shows normally.
> 
> If I add `rviz2` to the example, I can also run `rviz2` without problems.

You are right this also works for me. I manage some custom flakes via direnv and it seems like older artifacts somehow caused this problem. I removed the direnv cache and rebuilt these flakes. After that the problem has gone away.

okvik, 2025-09-16:
I too was doing some flake shenanigans with custom packages and overlays and systems and whatnot and using direnv, so maybe we both hit some weird corner case related to that. I can't reproduce it anymore in a more basic environment either so closing the issue.

closed 2025-09-16 by okvik