lopsided98/nix-ros-overlay #557: Unable to apply patches on ament vendor packages

KABILAN235, 2025-01-04:
## Problem
Applying patches on ament vendor packages through `patchAmentVendorGit` seems to be problematic especially in cases where a minor patch required on one of the vendor packages requires a version bump.

## Example

I'm using this package from the overlay that uses `patchAmentVendorGit` under the hood to fetch the source from [here](https://github.com/gazebosim/gz-sensors)

```nix
gz-sensors-vendor = lib.patchGzAmentVendorGit rosSuper.gz-sensors-vendor {
    version = "8.2.0";
    hash = "sha256-wEUJoHbvvImuFbaKk84maw5AoKhoEhdU0uOYVBtHhI0=";
  };
```

Problem arises when we want to include a patch like this [this](https://github.com/gazebosim/gz-sensors/commit/aaef365e6c51dcf98334e9f7a70e5fc4e3281f91.patch) have to be applied. Since  `patchAmentVendorGit` doesn't support patches, the solution here would be to bump up the version from `8.2.0 -> 8.2.1` which isin't ideal

## Proposed solution

Would be really cool if we can have a patches as an override attr which allows us to do something like this

```nix
gz-sensors-vendor = lib.patchGzAmentVendorGit rosSuper.gz-sensors-vendor {
    version = "8.2.0";
    hash = "sha256-wEUJoHbvvImuFbaKk84maw5AoKhoEhdU0uOYVBtHhI0=";
  }.overrideAttrs ({
    patches ? [], ...
  }: {
    patches = patches ++ [
      (self.fetchpatch {
        url = "https://github.com/gazebosim/gz-sensors/commit/aaef365e6c51dcf98334e9f7a70e5fc4e3281f91.patch";
        hash = "sha256-npFHdb0a3JBA8T6vke54DA08C93aNc/7c6xrfMBo7zI=";
        stripLen = 1;
      })
    ];
  });;
```

wentasah, 2025-01-04:
I don't understand what is the problem. Could you write which exact errors you encounter?

Isn't it sufficient to put `lib.patchGzAmentVendorGit` expression in parentheses before overriding it? Like this:
```nix
gz-sensors-vendor = (lib.patchGzAmentVendorGit rosSuper.gz-sensors-vendor {
    version = "8.2.0";
    hash = "sha256-wEUJoHbvvImuFbaKk84maw5AoKhoEhdU0uOYVBtHhI0=";
  }).overrideAttrs ({
    patches ? [], ...
  }: {
    patches = patches ++ [
      (self.fetchpatch {
        url = "https://github.com/gazebosim/gz-sensors/commit/aaef365e6c51dcf98334e9f7a70e5fc4e3281f91.patch";
        hash = "sha256-npFHdb0a3JBA8T6vke54DA08C93aNc/7c6xrfMBo7zI=";
        stripLen = 1;
      })
    ];
  });;
```

vic-bart, 2025-07-28:
I have run into the same issue; gz-msgs-vendor on the overlay is out-of-date and fails to build due to changes to protobuf.
I tried solving it as @KABILAN235 did:
```Nix
gz-msgs-vendor = (
  rosSelf.lib.patchGzAmentVendorGit rosSuper.gz-msgs-vendor {
    version = "11.1.0";
    hash = "sha256-M/rzUrL6uzpaRNLWJsGViY6Jk0bLtooEe+0eEEPS7PA=";
  }
);
```
But ran into the "mismatching versions with upstream" error (why am I prohibited from updating the vendor version?)
So I applied the patch as @wentasah suggested:
```Nix
gz-msgs-vendor = (rosSelf.lib.patchGzAmentVendorGit rosSuper.gz-msgs-vendor {
  version = "10.3.0";
  hash = "sha256-PQT8EpTxafldnKG3hDSXw2P22gLRg2EfMllrzaTaDEw=";
}).overrideAttrs ({
  nativeBuildInputs ? [],
  patches ? [], 
  ...
}: {
  nativeBuildInputs = nativeBuildInputs ++ [ self.breakpointHook ];
  patches = patches ++ [
    (self.fetchpatch {
      url = "https://github.com/gazebosim/gz-msgs/commit/22c57006798470db63e8ecaff7b49dce34d5e76f.patch";
      hash = "sha256-hItwP72gwm9GP4uqbjyhkXj/wTRZGxiVXKXhdTZJ4Dc=";
    })
  ];
});
```
But the patch cannot be applied as the target files do not seem to exist? 
(I'm not familiar with how building a vendor package differs from a standard package, but usually attaching a breakpoint has let me find the target files and correct issues with patches in the past)

[image]

[image]

wentasah, 2025-07-28:
Hi @vic-bart, I guess you are trying to build `gz-msgs-vendor` for `rolling` from the `master` branch of the overlay, which currently fails. Can you try switching to the `develop` branch of the overlay? This branch contains some fixes and building `gz-msgs-vendor` works for both `jazzy` and `rolling`. This can be easily verified by running:

    nix build github:lopsided98/nix-ros-overlay/develop#{jazzy,rolling}.gz-msgs-vendor

If you need something else, or encounter other issues, please write us the exact commands you're trying to run and the exact error messages you see. It is then much easier to diagnose the actual problem.

vic-bart, 2025-07-29:
Hi! I did try building gz-msgs-vendor on `develop` previously and it failed. The error was fixed in gz-msgs v11.1.0, but nix-ros-overlay `develop` is on v10.3.2. In the meantime, I've decided to fork nix-ros-overlay and update the necessary gz-*-vendor packages there.

closed 2026-02-15 by wentasah