lopsided98/nix-ros-overlay #560: nixos libsodium.so.26 when running gz example.

shot-codes, 2025-01-15:
I run into the following missing shared lib when trying to run `gz sim`  after `nix develop --impure github:lopsided98/nix-ros-overlay\#example-ros2-gz `

```
Library error for [/nix/store/3ld529mw7jarza951mzp5g9hyl5bha25-ros-jazzy-gz-sim-vendor-0.0.6-r1/lib64/libgz-sim8-gz.so.8.7.0]: libsodium.so.26: cannot open shared object file: No such file or directory
```

wentasah, 2025-01-19:
I cannot reproduce it. For me, `gz sim` works, but I end up with different hash of `libgz-sim8-gz.so.8.7.0`. Can you see libsodium in the output of the following command?

    ldd /nix/store/3ld529mw7jarza951mzp5g9hyl5bha25-ros-jazzy-gz-sim-vendor-0.0.6-r1/lib64/libgz-sim8-gz.so.8.7.0 | grep libsodium

shot-codes, 2025-01-21:
I just tried running it again and gz sim works! so a non issue I suppose. Thanks for the response!

closed 2025-01-21 by shot-codes