lopsided98/nix-ros-overlay #563: Rolling release `generated.nix` not complete

stevalkr, 2025-01-17:
Packages like `rclpy` and many more exist under `distros/rolling`, but not in `generated.nix`.

wentasah, 2025-01-17:
Yes. This happens when some dependency of a ROS package is not available. This is documented in [FAQ](https://github.com/lopsided98/nix-ros-overlay?tab=readme-ov-file#frequently-asked-questions) and every rosdistro sync PR contains a list of such packages. In this case it was ` python3-typing-extensions`. I've sent a [PR](https://github.com/ros/rosdistro/pull/44223) to fix that.

stevalkr, 2025-01-18:
Oops, I didn’t think about the possibility of basic things like rclpy not compiling. Sorry about that

closed 2025-01-18 by stevalkr

wentasah, 2025-01-31:
BTW, this will be fixed after merging #568 and updating pinned nixpkgs.