lopsided98/nix-ros-overlay #566: Why An Overlay?

AngleSideAngle, 2025-01-24:
Is there a design decision behind this project using an overlay to add ros packages to nixpkgs? From reading through the source code and examples, I couldn't find any indication that there's an advantage to distributing ros packages as an overlay, as opposed to having the package derivations in a separate flake from nixpkgs. Please let me know if I'm missing something...

Also, apologies if an issue wasn't the right way to ask this.

lopsided98, 2025-01-25:
There's no fundamental difference between an overlay and a flake. You can use this repo as a flake, or as a traditional overlay. The only thing really missing from the flake is support for flattening the package namespace so you don't have to use `legacyPackages`, but that doesn't matter much in practice.

lopsided98, 2025-01-25:
Flakes didn't exist when I started this project (or at least they weren't in common use), which explains why the flake support may seem a bit bolted on. I guess you could redesign it so that the packages were only available as flake outputs, but that would just limit possible use cases. I don't find flake package outputs particularly useful outside basic interactive usage. If I'm integrating a flake into a larger system I normally use the overlay output anyway.

AngleSideAngle, 2025-01-27:
I understand - My main reason for being curious was that the overlay seemed conceptually more complex than distributing the packages directly, but I didn't realize you could just use `nix-ros.rosPackages.distro.packagename`. Thanks for the response

closed 2025-01-27 by AngleSideAngle