lopsided98/nix-ros-overlay #567: How to override opencv with gtk2 support

iamthenoname, 2025-01-30:
I don't know if this is just due to my poor understanding of Nix, but trying to override opencv through nixpkgs.overlays doesn't work.

```
cv2.error: OpenCV(4.9.0) /build/source/modules/highgui/src/window.cpp:1272: error: (-2:Unspecified error) The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script in function 'cvShowImage'
```

The derivation for ros-env seems to depend on two different version of opencv... one of which wasn't built with gtk2 support. Maybe this is the culprit, but I don't know exactly what to do with this information.

```
$ nix-store --query --references /nix/store/5q0w0m94nba8hvv16iq7l9ymp950fyz1-ros-env.drv
...
/nix/store/6lpwvnrsi6qln936lhg9p2jylhnnrwnz-opencv-4.9.0.drv
/nix/store/cszbjmpmmz47jc023kdc4pfk272dqjg4-opencv-4.9.0.drv
...
```

```
$ nix-store --query --references /nix/store/6lpwvnrsi6qln936lhg9p2jylhnnrwnz-opencv-4.9.0.drv | grep gtk
$ nix-store --query --references /nix/store/cszbjmpmmz47jc023kdc4pfk272dqjg4-opencv-4.9.0.drv | grep gtk
/nix/store/z21scqp7mphyfhzwl52wnm1hn1dakm8n-gtk+-2.24.33.drv
```

Here is my `flake.nix`:
```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/develop";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";
  };
  outputs =
    {
      self,
      nix-ros-overlay,
      nixpkgs,
    }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [
            nix-ros-overlay.overlays.default
            (self: super: {
              opencv = super.opencv.override {
                enableGtk2 = true;
              };
            })
          ];
          config = {
            permittedInsecurePackages = [ "freeimage-unstable-2021-11-01" ];
          };
        };

        brisk = pkgs.stdenv.mkDerivation {
          pname = "brisk";
          version = "2.0.8";
          src = pkgs.fetchzip {
            url = "https://www.doc.ic.ac.uk/~sleutene/software/brisk-2.0.8.zip";
            hash = "sha256-pezKI15rqBpj0mhZDNEKTFkZ0WBe/2fyPEWcoMR+U0Q=";
          };
          nativeBuildInputs = with pkgs; [
            cmake
            opencv
          ];
        };

        rosbags = pkgs.python312Packages.buildPythonPackage rec {
          pname = "rosbags";
          version = "0.10.7";

          src = pkgs.fetchPypi {
            inherit pname version;
            hash = "sha256-PVH4S3rIt9QVflqlIHOgTXVK+lfIOLnWWO3XL84H+E0=";
          };

          pyproject = true;
          build-system = with pkgs.python312Packages; [
            setuptools
            setuptools-scm
            lz4
            numpy
            ruamel-yaml
            typing-extensions
            zstandard
          ];
        };
      in
      {
        devShells.default = pkgs.mkShell {
          name = "svin2";
          buildInputs = with pkgs; [
            boost186

            pcl
            glog
            gflags
            blas
            lapack
            eigen
            suitesparse
            pkgconf
            ceres-solver
            orocos-kdl

            brisk
          ];
          packages =  [
            pkgs.colcon
            pkgs.python312Packages.python-lsp-server
            pkgs.python312Packages.lz4
            pkgs.python312Packages.numpy
            pkgs.python312Packages.ruamel-yaml
            pkgs.python312Packages.typing-extensions
            pkgs.python312Packages.zstandard
            rosbags

            # ... other non-ROS packages
            (
              with pkgs.rosPackages.jazzy;
              buildEnv {
                paths = [
                  rviz2
                  ros-core
                  rosbag2
                  ament-cmake-core
                  python-cmake-module
                  pcl-ros
                  cv-bridge
                  tf2-sensor-msgs
                  nav-msgs
                  visualization-msgs
                  image-transport
                ];
              }
            )
          ];
        };
      }
    );
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
    max-cores = 2;
    jobs = 4;
  };
}
```

wentasah, 2025-01-31:
Try if overriding `opencv4` helps:
```nix
              opencv4 = super.opencv4.override {
                enableGtk2 = true;
              };

```

`opencv` is just an alias for `opencv4` and many packages depend directly on `opencv4` rather than on `opencv`.

wentasah, 2025-01-31:
Also, if you override `opencv` via nixpkgs overlay, it will result in rebuilding all packages depending on opencv and this will take long time. If you need gtk2 support just for the `brisk` package, you can override it just there:

```nix
        brisk = pkgs.stdenv.mkDerivation {
          pname = "brisk";
          version = "2.0.8";
          src = pkgs.fetchzip {
            url = "https://www.doc.ic.ac.uk/~sleutene/software/brisk-2.0.8.zip";
            hash = "sha256-pezKI15rqBpj0mhZDNEKTFkZ0WBe/2fyPEWcoMR+U0Q=";
          };
          nativeBuildInputs = with pkgs; [
            cmake
            (opencv.override { enableGtk2 = true; })
          ];
        };
```

iamthenoname, 2025-02-01:
Oh, wow. Replacing `opencv` with `opencv4` in the overlays really did work. I was overthinking this way too hard

Thanks for the tip about the `brisk` package, although I do need to have gtk support (during development, since I rely on `opencv`'s previews.) I'll just disable the overlay whenever I need to release the project.

Thank you very much!

closed 2025-02-01 by iamthenoname