lopsided98/nix-ros-overlay #576: Qt platform plugin not found for `joint-state-publisher-gui`

CTerasa, 2025-02-15:
# Description

Running the humble `joint_state_publisher_gui` I get the following error and the GUI does not start

```
$ ros2 run joint_state_publisher_gui joint_state_publisher_gui 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx.

[ros2run]: Aborted
```

Minimal `flake.nix` sample:

```
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs"; # IMPORTANT!!!
  };
  outputs =
    {
      self,
      nix-ros-overlay,
      nixpkgs,
    }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in
      {
        devShells.default = pkgs.mkShell {
          name = "Minimal sample for joint-state-publisher-gui";
          packages = [
            # ... other non-ROS packages
            (
              with pkgs.rosPackages.humble;
              buildEnv {
                paths =
                  [
                    ros-core
                    joint-state-publisher-gui
                  ];
              }
            )
          ];
        };
      }
    );
}
```

# Expected behavior

The GUI should start and the output should look similar to the following 

```
$ ros2 run joint_state_publisher_gui joint_state_publisher_gui 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[INFO] [1739624382.427781012] [joint_state_publisher]: Waiting for robot_description to be published on the robot_description topic...
[INFO] [1739624382.436617151] [joint_state_publisher]: Centering
```

# Observations and Workaround

It seems that the `ros2` execution does not have the correct `QT_PLUGINS_PATH` environment variable set. Thus the Qt QPA plugins cannot be found.

When I search for the Qt platform plugins and set the `QT_PLUGINS_PATH` the `joint_state_publisher_gui` starts up like expected.

```
$ export QT_PLUGIN_PATH=$(realpath $(dirname $(ls -d /nix/store/*/lib/qt-5.15.15/plugins/platforms/*xcb*| head -n1))/..)
$ ros2 run joint_state_publisher_gui joint_state_publisher_gui
```

With my limited knowledge of nix and nix-ros-overlay, I could not fix this in the package on my own.

I think the same issue prevents from starting the demos and working with the [Humble URDF tutorial](https://docs.ros.org/en/humble/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html) when using the `urdf-tutorial`.

wentasah, 2025-02-15:
#577 should fix this. Or you can add:
```nix
shellHook = ''
  export QT_PLUGIN_PATH=${pkgs.qt5.qtbase}/${pkgs.qt5.qtbase.qtPluginPrefix}
'';
```
as an argument to `pkgs.mkShell`.

CTerasa, 2025-02-16:
Thank you, the `shellHook` you proposed works like a charm.

Thought I tried something similar as in the change #577, however I had no success. Perhaps I made some kind of mistake on the way. I can verify that using the branch from #577 it is also working.
Keeping this issue open until #577 is merged.

Thanks and great support!

PR #577 (2025-02-17): joint-state-publisher-gui: Apply wrapQtAppsHook
Fixes #576