lopsided98/nix-ros-overlay #579: Unable to build ros-jazzy-gz-cmake-vendor: "Mismatch in version... fix in overrides.nix"

iamthenoname, 2025-02-19:
```shell
$ nix develop
error: builder for '/nix/store/zjzfwqjqv23lxxjic82y994zgzq04c8x-ros-jazzy-gz-cmake-vendor-0.0.9-r1.drv' failed with exit code 1;
       last 25 log lines:
       > fixing cmake files...
       > cmake flags: -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=OFF -DCMAKE_FIND_USE_PACKAGE_REGISTRY=OFF -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_BUILD_TYPE=Release -DBUILD_TESTING=OFF -DCMAKE_INSTALL_LOCALEDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/share/locale -DCMAKE_INSTALL_LIBEXECDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/libexec -DCMAKE_INSTALL_LIBDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/lib -DCMAKE_INSTALL_DOCDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/share/doc/gz_cmake_vendor -DCMAKE_INSTALL_INFODIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/share/info -DCMAKE_INSTALL_MANDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/share/man -DCMAKE_INSTALL_OLDINCLUDEDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/include -DCMAKE_INSTALL_INCLUDEDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/include -DCMAKE_INSTALL_SBINDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/sbin -DCMAKE_INSTALL_BINDIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/bin -DCMAKE_INSTALL_NAME_DIR=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1/lib -DCMAKE_POLICY_DEFAULT_CMP0025=NEW -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_STRIP=/nix/store/4ijy8jbsiqmj37avrk83gn2m903486mr-gcc-wrapper-14-20241116/bin/strip -DCMAKE_RANLIB=/nix/store/4ijy8jbsiqmj37avrk83gn2m903486mr-gcc-wrapper-14-20241116/bin/ranlib -DCMAKE_AR=/nix/store/4ijy8jbsiqmj37avrk83gn2m903486mr-gcc-wrapper-14-20241116/bin/ar -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -DCMAKE_INSTALL_PREFIX=/nix/store/z4bzmnwygkq8gwnbglsfip4y9wwc6m2f-ros-jazzy-gz-cmake-vendor-0.0.9-r1 -DAMENT_CMAKE_ENVIRONMENT_PARENT_PREFIX_PATH_GENERATION:BOOL=OFF -DPython3_EXECUTABLE:FILEPATH=/nix/store/0l539chjmcq5kdd43j6dgdjky4sjl7hl-python3-3.12.8/bin/python3.12 -DAMENT_CMAKE_ENVIRONMENT_PARENT_PREFIX_PATH_GENERATION:BOOL=OFF -DPython3_EXECUTABLE:FILEPATH=/nix/store/0l539chjmcq5kdd43j6dgdjky4sjl7hl-python3-3.12.8/bin/python3.12
       > -- The C compiler identification is GNU 14.2.1
       > -- The CXX compiler identification is GNU 14.2.1
       > -- Detecting C compiler ABI info
       > -- Detecting C compiler ABI info - done
       > -- Check for working C compiler: /nix/store/4ijy8jbsiqmj37avrk83gn2m903486mr-gcc-wrapper-14-20241116/bin/gcc - skipped
       > -- Detecting C compile features
       > -- Detecting C compile features - done
       > -- Detecting CXX compiler ABI info
       > -- Detecting CXX compiler ABI info - done
       > -- Check for working CXX compiler: /nix/store/4ijy8jbsiqmj37avrk83gn2m903486mr-gcc-wrapper-14-20241116/bin/g++ - skipped
       > -- Detecting CXX compile features
       > -- Detecting CXX compile features - done
       > -- Found ament_cmake_core: 2.5.3 (/nix/store/rccq6wjappjhxx8j8kv3aflg4npjm82a-ros-jazzy-ament-cmake-core-2.5.3-r1/share/ament_cmake_core/cmake)
       > -- Found Python3: /nix/store/0l539chjmcq5kdd43j6dgdjky4sjl7hl-python3-3.12.8/bin/python3.12 (found version "3.12.8") found components: Interpreter
       > -- Found ament_cmake_vendor_package: 2.5.3 (/nix/store/0v80kf0r7yqp2iwk1kg9w7wmhvjfwmw8-ros-jazzy-ament-cmake-vendor-package-2.5.3-r1/share/ament_cmake_vendor_package/cmake)
       > -- Found ament_cmake_export_dependencies: 2.5.3 (/nix/store/v272b87354g35wiixzzhinpd01sbs4fd-ros-jazzy-ament-cmake-export-dependencies-2.5.3-r1/share/ament_cmake_export_dependencies/cmake)
       > -- Found ament_cmake_test: 2.5.3 (/nix/store/klkr1i8m36hh42f2hn9gi4ilsf9x6zbb-ros-jazzy-ament-cmake-test-2.5.3-r1/share/ament_cmake_test/cmake)
       > CMake Error at CMakeLists.txt:100 (message):
       >   Mismatch in ros-jazzy-gz-cmake-vendor version (Nix: 3.5.3, upstream:
       >   3.5.4).  Fix this in overrides.nix.
       >
       >
       > -- Configuring incomplete, errors occurred!
       For full logs, run 'nix log /nix/store/zjzfwqjqv23lxxjic82y994zgzq04c8x-ros-jazzy-gz-cmake-vendor-0.0.9-r1.drv'.
error: 1 dependencies of derivation '/nix/store/agyik0f1cv4xn6nisnyhq6hqzz5csl05-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/0dzyjaia3zck32b5vfhl43rjs1mdn8xg-Example-project-env.drv' failed to build
```

My `flake.nix`:
```nix
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/develop";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "Example project";
          packages = [
            pkgs.colcon
            # ... other non-ROS packages
            (with pkgs.rosPackages.jazzy; buildEnv {
              paths = [
                ros-core
                ros2-control
                # ... other ROS packages
              ];
            })
          ];
        };
      });
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

wentasah, 2025-02-19:
#575 should fix that. Could you test it and report the result to the PR comment?

iamthenoname, 2025-02-19:
Can confirm that the PR resolved the issue for me. Thank you!

closed 2025-02-19 by iamthenoname