lopsided98/nix-ros-overlay #587: boost error when entering environment: type not declared in std

kostareg, 2025-03-01:
Hi, I'm trying to enter the dev environment for ROS2, I ran this command and it was compiling for 30-40 minutes before erroring out.

```
$ NIXPKGS_ALLOW_INSECURE=1 nix develop github:lopsided98/nix-ros-overlay#example-turtlebot3-gazebo --impure
do you want to allow configuration setting 'extra-substituters' to be set to 'https://ros.cachix.org' (y/N)? y
do you want to permanently mark this value as trusted (y/N)?
do you want to allow configuration setting 'extra-trusted-public-keys' to be set to 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' (y/N)? y
do you want to permanently mark this value as trusted (y/N)?
error: builder for '/nix/store/81yxlb2q7c5kl08vklk790k3j27w25qn-ros-noetic-tf-1.13.2-r1.drv' failed with exit code 2;
       last 25 log lines:
       > include/boost/math/tools/type_traits.hpp:281:12: error: 'remove_pointer_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:283:12: error: 'add_pointer_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:287:12: error: 'make_signed_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:289:12: error: 'make_unsigned_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:293:12: error: 'remove_extent_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:295:12: error: 'remove_all_extents_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:299:12: error: 'decay_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:301:12: error: 'enable_if_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:303:12: error: 'conditional_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:305:12: error: 'common_type_t' has not been declared in 'std'
       > include/boost/math/tools/type_traits.hpp:307:12: error: 'underlying_type_t' has not been declared in 'std'
       > In file included from include/boost/math/special_functions/math_fwd.hpp:28,
       >                  from include/boost/math/special_functions/round.hpp:20:
       > include/boost/math/tools/promotion.hpp:90:36: error: 'common_type_t' in namespace 'boost::math' does not name a template type; did you mean 'common_type'?
       > include/boost/math/tools/type_traits.hpp:402:60: error: 'is_final' is not a member of 'boost::math'; did you mean 'is_final_v'?
       > include/boost/math/tools/type_traits.hpp:402:70: error: expected primary-expression before '>' token
       > include/boost/math/tools/type_traits.hpp:402:73: error: '::value' has not been declared
       > In file included from include/boost/math/special_functions/math_fwd.hpp:28,
       >                  from include/boost/math/special_functions/round.hpp:20:
       > include/boost/math/tools/promotion.hpp:90:36: error: 'common_type_t' in namespace 'boost::math' does not name a template type; did you mean 'common_type'?
       > make[2]: *** [CMakeFiles/tf.dir/build.make:121: CMakeFiles/tf.dir/src/transform_listener.cpp.o] Error 1
       > make[2]: *** [CMakeFiles/tf.dir/build.make:93: CMakeFiles/tf.dir/src/tf.cpp.o] Error 1
       > make[2]: *** [CMakeFiles/tf.dir/build.make:107: CMakeFiles/tf.dir/src/transform_broadcaster.cpp.o] Error 1
       > make[1]: *** [CMakeFiles/Makefile2:2011: CMakeFiles/tf.dir/all] Error 2
       > make: *** [Makefile:136: all] Error 2
       For full logs, run 'nix log /nix/store/81yxlb2q7c5kl08vklk790k3j27w25qn-ros-noetic-tf-1.13.2-r1.drv'.
error: 1 dependencies of derivation '/nix/store/rbbkf9p21jvy6hf4ns2jr607m3j9vprs-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/d091ll7dzbqqc3av56c73bcvpyidab4h-nix-shell-env.drv' failed to build
```

[Full log](https://github.com/user-attachments/files/19036126/roslog.txt). Any advice appreciated - did I do the setup wrong?

wentasah, 2025-03-01:
Yes, this is a bug that should be fixed. But this is an example for ROS 1. If you want ROS 2, try `example-ros2-basic` or `example-ros2-gz`. These should work.

wentasah, 2026-08-29:
I think this can be closed.

closed 2026-08-29 by wentasah