lopsided98/nix-ros-overlay #588: Hash mismatch for examples-ros2-basic

kostareg, 2025-03-01:
```
❯ NIXPKGS_ALLOW_INSECURE=1 nix develop github:lopsided98/nix-ros-overlay#example-ros2-basic --impure
do you want to allow configuration setting 'extra-substituters' to be set to 'https://ros.cachix.org' (y/N)? y
do you want to permanently mark this value as trusted (y/N)?
do you want to allow configuration setting 'extra-trusted-public-keys' to be set to 'ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=' (y/N)? y
do you want to permanently mark this value as trusted (y/N)?
error: hash mismatch in fixed-output derivation '/nix/store/46h7cma84nw1l98d91ihqnxcjbpfqbld-2.6.9-1.tar.gz.drv':
         specified: sha256-BcuJFl70R2xZ5/VtZF3+EKSgOSS/IqowOO3lDRdPfdk=
            got:    sha256-tAfmAxVaLSvZv6lj4FLo3GLR/s1iYTFMQf6l3kfuVQ4=
error: 1 dependencies of derivation '/nix/store/ghnn28ar4ahg5c2rhdcakglqg9hgv51d-ros-humble-fastrtps-2.6.9-r1.drv' failed to build
error: 1 dependencies of derivation '/nix/store/cwqiz1czxnwlizc8wl70hpj1b7bz16rr-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/ikvh62mmq3zywjqm55zbggjqadvy49kl-nix-shell-env.drv' failed to build
```

kostareg, 2025-03-02:
Humble is broken, but jazzy works.

kostareg, 2025-03-02:
Overriding the hash fixes that issue, but then I get

```sh
warning: Git tree '/home/areg/projects/utfr-dv' is dirty
error: builder for '/nix/store/svs28xgv72x59l6nbwab3sp3x81h7ymv-ros-humble-builtin-interfaces-1.2.1-r1.drv' failed with exit code 2;
       last 25 log lines:
       > [ 66%] Built target builtin_interfaces__rosidl_typesupport_introspection_cpp
       > [ 69%] Linking CXX shared library libbuiltin_interfaces__rosidl_typesupport_cpp.so
       > [ 71%] Linking CXX shared library libbuiltin_interfaces__rosidl_typesupport_fastrtps_cpp.so
       > [ 71%] Built target builtin_interfaces__rosidl_typesupport_cpp
       > [ 71%] Built target builtin_interfaces__rosidl_typesupport_fastrtps_cpp
       > [ 74%] Linking CXX shared library libbuiltin_interfaces__rosidl_typesupport_fastrtps_c.so
       > [ 74%] Built target builtin_interfaces__rosidl_typesupport_fastrtps_c
       > [ 74%] Built target builtin_interfaces
       > [ 76%] Generating Python code for ROS interfaces
       > [ 76%] Built target builtin_interfaces__py
       > [ 79%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/rosidl_generator_py/builtin_interfaces/msg/_duration_s.c.o
       > [ 82%] Building C object CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/rosidl_generator_py/builtin_interfaces/msg/_time_s.c.o
       > /build/rcl_interfaces-release-release-humble-builtin_interfaces-1.2.1-1/build/rosidl_generator_py/builtin_interfaces/msg/_duration_s.c:11:10: fatal error: numpy/ndarrayobject.h: No such file or directory
       >    11 | #include "numpy/ndarrayobject.h"
       >       |          ^~~~~~~~~~~~~~~~~~~~~~~
       > compilation terminated.
       > make[2]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/build.make:76: CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/rosidl_generator_py/builtin_interfaces/msg/_duration_s.c.o] Error 1
       > make[2]: *** Waiting for unfinished jobs....
       > /build/rcl_interfaces-release-release-humble-builtin_interfaces-1.2.1-1/build/rosidl_generator_py/builtin_interfaces/msg/_time_s.c:11:10: fatal error: numpy/ndarrayobject.h: No such file or directory
       >    11 | #include "numpy/ndarrayobject.h"
       >       |          ^~~~~~~~~~~~~~~~~~~~~~~
       > compilation terminated.
       > make[2]: *** [CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/build.make:90: CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/rosidl_generator_py/builtin_interfaces/msg/_time_s.c.o] Error 1
       > make[1]: *** [CMakeFiles/Makefile2:475: CMakeFiles/builtin_interfaces__rosidl_generator_py.dir/all] Error 2
       > make: *** [Makefile:146: all] Error 2
       For full logs, run 'nix log /nix/store/svs28xgv72x59l6nbwab3sp3x81h7ymv-ros-humble-builtin-interfaces-1.2.1-r1.drv'.
error (ignored): error: cannot unlink '"/tmp/nix-build-ros-humble-std-srvs-4.2.4-r1.drv-5/build"': Directory not empty
error: 1 dependencies of derivation '/nix/store/49jsc25nnc8f24zl5xnfy2h5f92a64vz-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/yisd0gplkxk1cvlz259avbv5v0ys2kj9-Example-project-with-builtin_interfaces-fix-env.drv' failed to build
```

lopsided98, 2025-03-02:
I can't reproduce any of this.

That second issue was fixed a few weeks ago.

kostareg, 2025-03-02:
Can you reproduce the hash mismatch? Using [this](https://github.com/kostareg/nix-ros-overlay) fork (on master) works for me now.

wthomason-bdai, 2025-03-18:
I'm also seeing the same hash mismatch on humble, as of today.

wbthomason, 2025-03-18:
I'm seeing the same hash mismatch with humble on `master`, as of today.

wbthomason, 2025-03-18:
When I use your fork with the replaced hash, I do get past this error; unfortunately I then hit a compilation error with `octomap`. Not sure why, but it's probably a separate issue and maybe an indicator of a deeper problem with my flake.

wentasah, 2025-03-20:
The hash problem can be reproduced this way:
```
wget https://github.com/ros2-gbp/fastrtps-release/archive/release/humble/fastrtps/2.6.9-1.tar.gz -O wget.tgz && nix hash file wget.tgz
sha256-tAfmAxVaLSvZv6lj4FLo3GLR/s1iYTFMQf6l3kfuVQ4=
```

diffoscope reveals the following:
[image]

That's because https://github.com/ros2-gbp/fastrtps-release gets redirected to https://github.com/ros2-gbp/fastdds-release.

lopsided98, 2025-03-20:
Not helpful, but they really didn't do a great job with that renaming...

wentasah, 2025-03-20:
I think this is now fixed in `develop`, because it refers to `fastdds`. Master is broken, but I cannot reproduce it, because I have the old tarball in the nix store.