lopsided98/nix-ros-overlay #598: ros-jazzy-libcamera: Dependency "lttng-ust" not found

iamthenoname, 2025-03-16:
Not sure why `lttng-ust` is required, since it should only be optional

Log output:
```
Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/z430yjac0qfj9b7l4qbcq72z6rmw6mvz-0.4.0-1.tar.gz
source root is libcamera-release-release-jazzy-libcamera-0.4.0-1
setting SOURCE_DATE_EPOCH to timestamp 1735822795 of file "libcamera-release-release-jazzy-libcamera-0.4.0-1/utils/update-mojo.sh"
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
mesonConfigurePhase flags: --prefix=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1 --libdir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/lib --libexecdir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/libexec --bindir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/bin --sbindir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/sbin --includedir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/include --mandir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/share/man --infodir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/share/info --localedir=/nix/store/i1g48drr7skbba13slhizjycnhz9627f-ros-jazzy-libcamera-0.4.0-r1/share/locale -Dauto_features=enabled -Dwrap_mode=nodownload --buildtype=plain
The Meson build system
Version: 1.7.0
Source dir: /build/libcamera-release-release-jazzy-libcamera-0.4.0-1
Build dir: /build/libcamera-release-release-jazzy-libcamera-0.4.0-1/build
Build type: native build
Project name: libcamera
Project version: 0.4.0
C compiler for the host machine: gcc (gcc 14.2.1 "gcc (GCC) 14.2.1 20241116")
C linker for the host machine: gcc ld.bfd 2.43.1
C++ compiler for the host machine: g++ (gcc 14.2.1 "g++ (GCC) 14.2.1 20241116")
C++ linker for the host machine: g++ ld.bfd 2.43.1
Host machine cpu family: x86_64
Host machine cpu: x86_64
Header "fcntl.h" has symbol "F_ADD_SEALS" : YES 
Header "unistd.h" has symbol "issetugid" : NO 
Header "locale.h" has symbol "locale_t" : YES 
Header "sys/mman.h" has symbol "memfd_create" : YES 
Header "stdlib.h" has symbol "secure_getenv" : YES 
Compiler for C supports arguments -Wno-c99-designator: NO 
Found pkg-config: YES (/nix/store/8cczxvv6hpngc9mxal0wx0lzny4vjp4n-pkg-config-wrapper-0.29.2/bin/pkg-config) 0.29.2
Did not find CMake 'cmake'
Found CMake: NO
Run-time dependency lttng-ust found: NO (tried pkgconfig)

A full log can be found at /build/libcamera-release-release-jazzy-libcamera-0.4.0-1/build/meson-logs/meson-log.txt
```

wentasah, 2025-03-16:
The reason is that libcamera is built by Meson and nixpkgs support for meson enables all optional features by default. We can either provide the missing dependencies, or disable the feature for which dependencies are missing. In #600 I implemented the latter together with other fixes that were required. Let me know whether the PR works for you.

iamthenoname, 2025-03-17:
Yeah, works for me. I don't think it's necessary to include `lttng-ust`

PR #600 (2025-03-27): libcamera: Fix builds for jazzy and rolling
This fixes libcamera builds for jazzy and rolling. The version for humble is still broken, because its build system depends on Python module imp, which was removed from Python 3.12.

Fixes #598
closed 2025-03-17 by iamthenoname