lopsided98/nix-ros-overlay #601: Environment variables missing in newer version

stevalkr, 2025-03-18:
I updated my environment and discovered that certain environment variables were missing, such as `$ROS_PACKAGE_PATH` was empty and `$CMAKE_PREFIX_PATH` was not set. Instead I found `$NIXPKGS_CMAKE_PREFIX_PATH`. However, this change causes `pluginlib`’s library finding strategy to malfunction, as it relies on `$CMAKE_PREFIX_PATH`. 

flake.nix:

```nix
rec {
  description = "Nix Environment";

  inputs = {
    ros-overlay.url = "github:lopsided98/nix-ros-overlay/develop";
    nixpkgs.follows = "ros-overlay/nixpkgs";
    flake-utils.follows = "ros-overlay/flake-utils";
  };

  outputs = { self, nixpkgs, ros-overlay, flake-utils }:
    flake-utils.lib.eachDefaultSystem
      (system:
        let
          pkgs = import nixpkgs {
            inherit system;
            overlays = [ self.overlays.default ];
          };
        in
        rec {
          devShell = pkgs.mkShell {
            buildInputs = [
              pkgs.rosPackages.noetic.rosbag-storage
            ];
          };
        }
      )
    // {
      overlays.default = nixpkgs.lib.composeExtensions ros-overlay.overlays.default (
        final: prev: {
          # clang 19 won't compile without the patch
          poco = prev.callPackage ./pkgs/poco.nix { };
          rosPackages.noetic = prev.rosPackages.noetic.overrideScope (
            (ros-final: ros-prev: {
              rosbag-storage = ros-prev.rosbag-storage.overrideAttrs (finalAttrs: prevAttrs: {
                patches = (prevAttrs.patches or []) ++ [ ./pkgs/rosbag-storage/fix_unsigned_char.patch ];
              });
            })
          );
        }
      );
    };
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

flake.lock (old):
```json
{
  "nodes": {
    "flake-utils": {
      "inputs": {
        "systems": "systems"
      },
      "locked": {
        "lastModified": 1705309234,
        "narHash": "sha256-uNRRNRKmJyCRC/8y1RqBkqWBLM034y4qN7EprSdmgyA=",
        "owner": "numtide",
        "repo": "flake-utils",
        "rev": "1ef2e671c3b0c19053962c07dbda38332dcebf26",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "flake-utils",
        "type": "github"
      }
    },
    "nixpkgs": {
      "locked": {
        "lastModified": 1707268954,
        "narHash": "sha256-2en1kvde3cJVc3ZnTy8QeD2oKcseLFjYPLKhIGDanQ0=",
        "owner": "lopsided98",
        "repo": "nixpkgs",
        "rev": "f8e2ebd66d097614d51a56a755450d4ae1632df1",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "nix-ros",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "root": {
      "inputs": {
        "flake-utils": [
          "ros-overlay",
          "flake-utils"
        ],
        "nixpkgs": [
          "ros-overlay",
          "nixpkgs"
        ],
        "ros-overlay": "ros-overlay"
      }
    },
    "ros-overlay": {
      "inputs": {
        "flake-utils": "flake-utils",
        "nixpkgs": "nixpkgs"
      },
      "locked": {
        "lastModified": 1724117129,
        "narHash": "sha256-syYDeNi50lYGaSajggdgcLbV8P/E3v77F2fF5vTuCoI=",
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "rev": "1992646cf65ac8f036a21a2005e10c4454c9e3fc",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "develop",
        "repo": "nix-ros-overlay",
        "type": "github"
      }
    },
    "systems": {
      "locked": {
        "lastModified": 1681028828,
        "narHash": "sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768=",
        "owner": "nix-systems",
        "repo": "default",
        "rev": "da67096a3b9bf56a91d16901293e51ba5b49a27e",
        "type": "github"
      },
      "original": {
        "owner": "nix-systems",
        "repo": "default",
        "type": "github"
      }
    }
  },
  "root": "root",
  "version": 7
}
```

flake.lock (new):
```json
{
  "nodes": {
    "flake-utils": {
      "inputs": {
        "systems": "systems"
      },
      "locked": {
        "lastModified": 1731533236,
        "narHash": "sha256-l0KFg5HjrsfsO/JpG+r7fRrqm12kzFHyUHqHCVpMMbI=",
        "owner": "numtide",
        "repo": "flake-utils",
        "rev": "11707dc2f618dd54ca8739b309ec4fc024de578b",
        "type": "github"
      },
      "original": {
        "owner": "numtide",
        "repo": "flake-utils",
        "type": "github"
      }
    },
    "nixpkgs": {
      "locked": {
        "lastModified": 1739580444,
        "narHash": "sha256-+/bSz4EAVbqz8/HsIGLroF8aNaO8bLRL7WfACN+24g4=",
        "owner": "lopsided98",
        "repo": "nixpkgs",
        "rev": "8bb37161a0488b89830168b81c48aed11569cb93",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "nix-ros",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
    "root": {
      "inputs": {
        "flake-utils": [
          "ros-overlay",
          "flake-utils"
        ],
        "nixpkgs": [
          "ros-overlay",
          "nixpkgs"
        ],
        "ros-overlay": "ros-overlay"
      }
    },
    "ros-overlay": {
      "inputs": {
        "flake-utils": "flake-utils",
        "nixpkgs": "nixpkgs"
      },
      "locked": {
        "lastModified": 1742152777,
        "narHash": "sha256-OkQcq3eBWLuJ9I7lNSD3Y0MWm+Cn8FGLiqx5aDkf5r0=",
        "owner": "lopsided98",
        "repo": "nix-ros-overlay",
        "rev": "e99617dda797c3f46c7bde8880e54c759230efd2",
        "type": "github"
      },
      "original": {
        "owner": "lopsided98",
        "ref": "develop",
        "repo": "nix-ros-overlay",
        "type": "github"
      }
    },
    "systems": {
      "locked": {
        "lastModified": 1681028828,
        "narHash": "sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768=",
        "owner": "nix-systems",
        "repo": "default",
        "rev": "da67096a3b9bf56a91d16901293e51ba5b49a27e",
        "type": "github"
      },
      "original": {
        "owner": "nix-systems",
        "repo": "default",
        "type": "github"
      }
    }
  },
  "root": "root",
  "version": 7
}
```

wentasah, 2025-03-18:
Wasn't this fixed in #500? If not, I'll be able to look at this later.

stevalkr, 2025-03-18:
I thought this was the same issue but it turns out this actually contains two problems:

1. The `pluginlib` library explicitly searches for `$CMAKE_PREFIX_PATH` at runtime, and the patch on `catkin` is unlikely to work.
https://github.com/ros/pluginlib/blob/6e2a8295e0a8a92259577a2b29909738e210612c/pluginlib/include/pluginlib/class_loader_imp.hpp#L309-L330

2. The variable `$ROS_PACKAGE_PATH` is empty

stevalkr, 2025-03-18:
The issue with the `$ROS_PACKAGE_PATH` variable occurred after the commit https://github.com/lopsided98/nix-ros-overlay/commit/b6676bf7dc1b03afe9582c5fd99025b3a4b7f4f9. With the nixpkgs revision "9f4128e00b0ae8ec65918efeba59db998750ead6," the `$ROS_PACKAGE_PATH` is correctly set. However, with the revision "345c263f2f53a3710abe117f28a5cb86d0ba4059," it is not. The `catkin` patch is manually applied during compilation.

ultra-azu, 2025-09-14:
I'm currently having this issue, has anyone found a workaround?

wentasah, 2026-08-29:
Closing as this seems to be related to Noetic, which is EOL.

closed 2026-08-29 by wentasah