lopsided98/nix-ros-overlay #604: Hash mismatch when following flake instructions in README

ryleu, 2025-03-26:
Followed the instructions in the README to a T, but I did not get a working dev environment: 

[image]

```bash
$ nix flake init --template github:lopsided98/nix-ros-overlay
wrote: /home/ryleu/Documents/astra/rover-Basestation-Release/flake.nix

$ nix develop
warning: Git tree '/home/ryleu/Documents/astra/rover-Basestation-Release' is dirty
warning: creating lock file '/home/ryleu/Documents/astra/rover-Basestation-Release/flake.lock': 
• Added input 'nix-ros-overlay':
    'github:lopsided98/nix-ros-overlay/c3e6e964fcdd917193dca68cc5cb16f128fe5e1b?narHash=sha256-TzKPxouvrol38ng/VmHkNxObdon0taXHhcPmo/09cJo%3D' (2024-11-16)
• Added input 'nix-ros-overlay/flake-utils':
    'github:numtide/flake-utils/c1dfcf08411b08f6b8615f7d8971a2bfa81d5e8a?narHash=sha256-X6rJYSESBVr3hBoH0WbKE5KvhPU5bloyZ2L4K60/fPQ%3D' (2024-09-17)
• Added input 'nix-ros-overlay/flake-utils/systems':
    'github:nix-systems/default/da67096a3b9bf56a91d16901293e51ba5b49a27e?narHash=sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768%3D' (2023-04-09)
• Added input 'nix-ros-overlay/nixpkgs':
    'github:lopsided98/nixpkgs/345c263f2f53a3710abe117f28a5cb86d0ba4059?narHash=sha256-sjiGsMh%2B1cWXb53Tecsm4skyFNag33GPbVgCdfj3n9I%3D' (2024-09-13)
• Added input 'nixpkgs':
    follows 'nix-ros-overlay/nixpkgs'
warning: Git tree '/home/ryleu/Documents/astra/rover-Basestation-Release' is dirty
error: hash mismatch in fixed-output derivation '/nix/store/jpd2m9415299inb6c3843bcfflmg0l33-2.6.8-1.tar.gz.drv':
         specified: sha256-fTEJWBydIpOorHLuHykwXdCHOXp3u0gPH3mlXIBDQ6w=
            got:    sha256-ngNzOWIqCaaJF8OXdbrP/ig9yIa7+XhcJznrg3hqAK8=
error: 1 dependencies of derivation '/nix/store/dg95z11msv743n8gi9x1yxwnjyw16hai-ros-humble-fastrtps-2.6.8-r1.drv' failed to build
error (ignored): error: cannot unlink '"/tmp/nix-build-ros-humble-ament-index-cpp-1.4.0-r2.drv-0/build/ament_index-release-release-humble-ament_index_cpp-1.4.0-2/build/CMakeFiles"': Directory not empty
error: 1 dependencies of derivation '/nix/store/zz8n3a0i574qqywc6y5rdsi1nzdyqmwm-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/ks61xxb20073vmgszyizz7xg58yccrvn-Example-project-env.drv' failed to build
```

ryleu, 2025-03-26:
I'm guessing things just need to be updated upstream, because I'm able to get it working fine with the flake.nix in the gazebo example you made.

wentasah, 2025-03-27:
This is caused by #588. Since the master branch was updated today, I think that if you run `nix flake update` and try again, it should work. Please let us know if it is the case.

ryleu, 2025-03-27:
I'll let you know in a bit once I get back. Would the fix be as simple as adding a cron schedule to the build action? I can make a PR for that if you want.

ryleu, 2025-03-27:
[image]

Still doesn't seem to be working.

ryleu, 2025-03-27:
After removing my flake.lock, it works fine. Thanks!

closed 2025-03-27 by ryleu