lopsided98/nix-ros-overlay #608: Gazebo Nix

blaggacao, 2025-03-30:
- https://github.com/PX4/PX4-Autopilot/pull/24635/commits (second commit)

I'm not really sure if I'm doing the right thing there.

I used an LLM to automatically translate the dependency output of `colcon info` into a nix derivations.

Some things started to build successfully.

How can I now assemble this into Gazebo? I see https://github.com/lopsided98/nix-ros-overlay/blob/develop/pkgs/gazebo/default.nix, but I wonder if all the other `gz-*` that I had packaged are missing there. Or is this a question of widely different versions with widely different build layouts?

blaggacao, 2025-03-30:
PS: can we work together on the gazebo packaging, if interest there be?

wentasah, 2025-04-02:
Hi, to package 3rd party ROS packages, you can use [ros2nix](https://github.com/wentasah/ros2nix/). I'd trust it more than a LLM :-)

Regarding gazebo. I myself don't use it, but as far as I know, recent ROS 2 distributions include vendored versions of Gazebo that are meant to work with that distribution. These are named `gz-*-vendor` and they more or less work (if you ignore insecure freeimage). For example #591 fixed some nav2 packages which were needed for running nav2 demos, which use gazebo.

The version you point to is an old (after EOL) version of gazebo. I assume it would be removed after ROS Noetic goes EOL, which should be next month.

The `gz-*` packages are a part of different ROS distributions, which are available under `rosPackages.<distro>` attributes. Find them for example [here](https://hydra.iid.ciirc.cvut.cz/jobset/nix-ros-overlay/develop-all-x86_64#tabs-jobs).

wentasah, 2025-04-02:
ros2nix is meant to be used with this overlay. It creates derivations by calling `buildRosPackage`, which calls either `stdenv.mkDerivation` or `buildPythonPackage` depending on package type. If you want to be independent of this overlay, you would need to change it to `mkDerivation` manually. Also, sometimes, other manual fixes are needed. But yes, you could have saved some time by using it.

muellerbernd, 2025-05-13:
@blaggacao If you only need gazebo without ROS you can also take a look into [my gazebo-sim-overlay](https://github.com/muellerbernd/gazebo-sim-overlay)

wentasah, 2026-08-29:
I'm closing this. We have `gz-*-vendor` packages working in this overlay and there are multiple other Nix repos where Gazebo works.

closed 2026-08-29 by wentasah