lopsided98/nix-ros-overlay #615: How to setup a worspace dev environment

tomkimsour, 2025-04-09:
I am trying to setup a developer environment using nix. But I seem to struggle with building the workspace and sourcing it. I was wondering if you had any recommandation on how to setup an envrionment where I could make Clang work, compile my packages and source them to test them.

wentasah, 2025-04-09:
I think you can start from the [template mentioned in the README](https://github.com/lopsided98/nix-ros-overlay#flakes). Add to it whatever packages you need (clang etc.). Enter the environment by running `nix develop` (you need enable `--experimental-features 'nix-command flakes'`) and you should be able to compile and source your workspace as usual. If not, write here and we'll try to help.
 
If this works and you don't want to run `nix develop` every time you want to work on the project, you can automate entering the environment by using [direnv](https://direnv.net/), optionally together with [ros-direnv](https://github.com/wentasah/ros-direnv).

tomkimsour, 2025-04-09:
I'm testing with noetic right now. I am able to run `nix develop` and add the packages I want from the noetic distribution.

However I'm trying to reproduce the workflow I have with workspaces. Which is basically dumping the various packages I am working with, which have inter dependencies usually (like ros package containing only ros msgs), then I usually do a caktin/colcon build, source (to add the python paths for my editor) and then enter my editor.

Also, I want the compile_commands.json to be generated from the build of the workspace for clangd to work.

If you have examples on how you develop packages in a workspace for packages which are not from the classic rosdistro, I'd be curious to have them.

wentasah, 2025-04-09:
This is exactly what I'm doing and it works with the provided template. The only difference from the traditional (non-Nix) ROS workflow is that you don't source `/opt/ros/.../setup.bash`, but use `nix develop` instead. The rest should work the same. I don't use noetic (much), so don't know all the tricks, but for ROS 2, if you want to have `compile_commands.json`, just run `colcon build --cmake-args -DCMAKE_EXPORT_COMPILE_COMMANDS=ON`.

What might be your problem is that you're using `clangd` from your host distribution. This won't work, because Nix-provided compilers use variables like `NIX_CFLAGS_COMPILE` to search for header files and libraries and host's `clangd` doesn't know about it. So you need to add `pkgs.clangd` to your `devShell` in `flake.nix` and then it should work.

tomkimsour, 2025-04-10:
I got it to work the way you describe it. However I'm finding the setup bothersome to work with. I'm not sure this workflow is sustainable for me unfortunatly. Too much overhead of adding the packages I need one by one into my flake.nix . Seems like it's missing a metapkg for devs.
I had to change the path of my clangd to the nix one manually, which might change if I use a different environment. I could make it generic I think but it's additional overhead.

It feels like if you want to use nix it's all or nothing. I'm using nvim without home-manager, which I guess would be better integrated.

Thanks for your help setting up the workspace tho ! The work you're doing is quite cool ! Talking from as old ros-gentoo maintainer.

wentasah, 2025-04-10:
> I got it to work the way you describe it. However I'm finding the setup bothersome to work with. I'm not sure this workflow is sustainable for me unfortunatly. Too much overhead of adding the packages I need one by one into my flake.nix. Seems like it's missing a metapkg for devs.

Yeah, this overlay contains just official ROS packages with metapackages like `ros-core` or `ros2cli_common_extensions`. Anything else has to be done manually. However, when thinking about it, I could probably extend `ros2nix` to automatically generate development shell for all packages in the workspace. This should be pretty straightforward.

> I had to change the path of my clangd to the nix one manually, which might change if I use a different environment. I could make it generic I think but it's additional overhead.

clangd from Nix should be in your `$PATH`. So if your editor runs just `clangd` without an absolute path, it should work well. This is how I have it configured.
 
> It feels like if you want to use nix it's all or nothing. I'm using nvim without home-manager, which I guess would be better integrated.

Yes and no. One goal of Nix is that it gives you reproducible setup, which works exactly the same independently of the host system. If you want to mix, e.g., Nix-provided libraries and host libraries, the result would be very fragile. So you have to draw the separation line somewhere. Therefore compilers and similar things have to be from Nix, but other tools like editors can be from outside. The only requirement is that the outside tools respect environment variables set by Nix like PATH, LD_LIBRARY_PATH etc.