lopsided98/nix-ros-overlay #617: file does not exist in binary cache + noob

comejv, 2025-04-11:
Hello,
I'm quite recent to nix and especially flakes and the flake I have for ROS development is giving me troubles :
It used to work but after a `nix-collect-gargbages` and a `nix-channel --update` it suddenly started giving me this error when I develop it :
```
error: file 'nar/45f9469aa4d6b85ba4c92d9c69f391b3f63f936b298b0fa7e95cc29c5c189f86.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/7e5cb325156b86a253a4b9b9d59f951632aa76c82cf08a0edbeac28cb6d60350.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/c64bdf685dd6e57327d87c46132bf111cd08082cd6b57973ed6d45e1c15b6485.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/3bae09c8f7a1526dab9abb5189cfc2c87d49c29bf360cfdfe0f16f995f0c6043.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/4695a6c3d365f4fe1695f97ce54c2fb3a079cf392f236ab4be49c345f9f82851.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/322abd407d7cc10f21c38a8c6ef68918b88bd489c471ab09b7dce94e804c766a.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/fd6e5b9b4dc224ed78f31979ad7e805c6264f231fad13ad5f232d8ae99b77c13.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/4bfa3dcf596f4e8c065d96fafa5b18cfd60a2fc14c1d0b6c753df46733848791.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/4b318f306ab3d8d4e96442df8cd913fa00091afbcb85810172f15d59a7b11a55.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
error: file 'nar/c75b47f6b9c82e2164c33267abb78da743622ab2c4c4f89d5c4a9cf9fea872f1.nar.zst' does not exist in binary cache 'https://ros.cachix.org'
```

And it goes on and on and then tries to build everything from source. 

Here's my flake :
```
{
  description = "BeamNG ROS 2 Development Environment (Fish Only)";

  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/master";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";
  };

  outputs =
    {
      self,
      nix-ros-overlay,
      nixpkgs,
    }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
        rosDistro = pkgs.rosPackages.humble;

        # Define the custom python environment with beamngpy
        pythonWithBeamNGpy = pkgs.python3.withPackages (
          ps: with ps; [
            numpy
            msgpack
            pillow
            jinja2
            scipy
            matplotlib
            seaborn
            ipython
            black
            flake8
            mypy
            pytest
            pydocstyle
            # Build beamngpy from PyPI source
            (buildPythonPackage rec {
              pname = "beamngpy";
              version = "1.30";
              src = pkgs.fetchPypi {
                inherit pname version;
                sha256 = "sha256-2lsuxJ6zCx2jFEx3t7UbT6RORd7D82YjK7jIk0tfD0w=";
              };
              # Dependencies needed by beamngpy at runtime
              propagatedBuildInputs = [
                numpy
                msgpack
                pillow
                jinja2
                scipy
                matplotlib
                seaborn
              ];
              doCheck = false;
            })
          ]
        );

        # Combine ROS packages into a single environment derivation
        rosEnv = rosDistro.buildEnv {
          paths = with rosDistro; [
            ros-core
            ros-base
            # Select the desired RMW implementation explicitly
            rmw-fastrtps-cpp
            # Dependencies for rmw_fastrtps_cpp
            rmw-fastrtps-shared-cpp
            fastrtps
            fastcdr
            # Core build/ament tools
            ament-cmake-core
            ament-lint-auto
            ament-lint-common
            python-cmake-module
            # User requested tools
            plotjuggler
            plotjuggler-ros
          ];
        };

      in
      {
        devShells.default = pkgs.mkShell {
          name = "bng-ros-dev-fish";

          packages = [
            # Build tools
            pkgs.colcon
            pkgs.cmake
            pkgs.ninja
            pkgs.pkg-config
            pkgs.gnumake
            pkgs.gcc
            pkgs.gdb
            pkgs.binutils

            # Custom Python environment
            pythonWithBeamNGpy

            # Python build/lint tools (available in the shell PATH)
            pkgs.pyright
            pkgs.python3.pkgs.setuptools # Often needed by colcon/build systems
            pkgs.python3.pkgs.wheel # Often needed by colcon/build systems
            pkgs.python3.pkgs.pip
            pkgs.python3.pkgs.cython # If needed for building certain packages

            # ROS environment
            rosEnv

            # Graphics/Simulation deps
            pkgs.libGL
            pkgs.libGLU
            pkgs.glfw
            pkgs.qt5.qtbase

            # Fish plugin needed for the 'cb' function sourcing bash env
            pkgs.fishPlugins.bass
          ];

          # Environment variables set directly by mkShell
          ROS_DOMAIN_ID = "42";
          RMW_IMPLEMENTATION = "rmw_fastrtps_cpp";

          # shellHook runs in bash/sh before fish starts
          # Use POSIX syntax here
          shellHook = ''
            # Unset PYTHONHOME/PYTHONPATH if they exist, ROS/colcon sets its own
            unset PYTHONHOME
            unset PYTHONPATH

            # Add ROS paths
            export AMENT_PREFIX_PATH="${rosEnv}:$AMENT_PREFIX_PATH"
            export CMAKE_PREFIX_PATH="${rosEnv}:$CMAKE_PREFIX_PATH"

            # Add custom Python environment's bin to PATH and site-packages maybe needed
            export PATH="${pythonWithBeamNGpy}/bin:$PATH"

            # Add libgl for graphic support
            export LD_LIBRARY_PATH=${pkgs.libGL}/lib:${pkgs.libGLU}/lib:$LD_LIBRARY_PATH

            # Python headers
            export CFLAGS="-I${pythonWithBeamNGpy}/include/python${pythonWithBeamNGpy.pythonVersion}"
            echo "Entering ROS2 + BeamNG Nix shell..."
          '';

          interactiveShellInit = ''
            # # Define the 'cb' function for fish using bass
            # function cb --description 'Build with colcon and source install/setup.fish'
            #   colcon build --symlink-install $argv
            #   if test -f install/setup.bash
            #       bass source install/setup.bash
            #       echo "? Built and sourced install/setup.bash using bass"
            #   else
            #       echo "?? Build finished, but no install/setup.bash found to source."
            #   end
            # end

            echo "ROS2 + BeamNG development environment (fish) activated."
            echo "Run 'cb' to build and source."
          '';
        };
      }
    );

  # Configure Nix to use the ROS binary cache
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

I am using fish as my shell (not login) and I also have issues because the interactive shell init is not being run, and when it is there are apparently syntax errors in the cb function definition.

Can you help me?

Thank you.

wentasah, 2025-04-11:
I think that the `error: ... does not exist in binary cache 'https://ros.cachix.org'` must be some temporary network or Cachix error. When I try your flake, `nix develop` works correctly. But things seem to build from source even here. It seems that the github action responsible for populating the binary cache [has some problems](https://github.com/lopsided98/nix-ros-overlay/actions/runs/14097036509). I'm not much familiar with that code. Maybe @lopsided98 can tell us more.

If you want, you can try my [experimental binary cache](https://github.com/wentasah/nix-ros-hydra?tab=readme-ov-file#binary-cache). Hopefully, it will work for you. If not, let me know.

Regarding `interactiveShellInit`, I don't see it anywhere in the documentation [here](https://nixos.org/manual/nixpkgs/unstable/#sec-pkgs-mkShell) or [here](https://nix.dev/manual/nix/2.28/command-ref/nix-shell) so it probably doesn't exist. Just define your functions in `shellHook`. However, these will only work in the bash shell spawned by `nix develop`, not in fish or whatever other shell you might run from there.

If you just want to automatically source `install/setup.bash`, my [ros-direnv](https://github.com/wentasah/ros-direnv) might help you even with fish.

comejv, 2025-04-11:
That's a lot of useful info, thank you !
How would I go about using your experimental cache?

wentasah, 2025-04-11:
Replace `nixConfig` in your `flake.nix`.

closed 2025-04-11 by comejv