lopsided98/nix-ros-overlay #621: pkgs.rosPackages.humble.cartographer-ros: Build fail

pjbuterbaugh, 2025-04-17:
`cartographer-ros` under the `humble` distribution fails to build when declared. See a truncated log snippet below:

```
building '/nix/store/kvfyf4qdw4nvfb7fb35fnrkdpai0n2gw-ros-humble-cartographer-ros-2.0.9002-r1.drv'...
error: builder for '/nix/store/kvfyf4qdw4nvfb7fb35fnrkdpai0n2gw-ros-humble-cartographer-ros-2.0.9002-r1.drv' failed with exit code 2;
       last 25 log lines:
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:171:24: error: expected ';' at end of member declaration
       >   171 |       int trajectory_id) EXCLUSIVE_LOCKS_REQUIRED(mutex_);
       >       |                        ^
       >       |                         ;
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:171:51: error: 'mutex_' has not been declared
       >   171 |       int trajectory_id) EXCLUSIVE_LOCKS_REQUIRED(mutex_);
       >       |                                                   ^~~~~~
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:171:26: error: ISO C++ forbids declaration of 'EXCLUSIVE_LOCKS_REQUIRED' with no type []
       >   171 |       int trajectory_id) EXCLUSIVE_LOCKS_REQUIRED(mutex_);
       >       |                          ^~~~~~~~~~~~~~~~~~~~~~~~
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:186:37: error: expected ';' at end of member declaration
       >   186 |   std::shared_ptr<MapBuilderBridge> map_builder_bridge_ GUARDED_BY(mutex_);
       >       |                                     ^~~~~~~~~~~~~~~~~~~
       >       |                                                        ;
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:186:68: error: 'mutex_' is not a type
       >   186 |   std::shared_ptr<MapBuilderBridge> map_builder_bridge_ GUARDED_BY(mutex_);
       >       |                                                                    ^~~~~~
       > /build/cartographer_ros-release-release-humble-cartographer_ros-2.0.9002-1/include/cartographer_ros/node.h:186:57: error: ISO C++ forbids declaration of 'GUARDED_BY' with no type []
       >   186 |   std::shared_ptr<MapBuilderBridge> map_builder_bridge_ GUARDED_BY(mutex_);
       >       |                                                         ^~~~~~~~~~
       > make[2]: *** [CMakeFiles/cartographer_ros.dir/build.make:93: CMakeFiles/cartographer_ros.dir/src/map_builder_bridge.cpp.o] Error 1
       > make[2]: *** [CMakeFiles/cartographer_ros.dir/build.make:163: CMakeFiles/cartographer_ros.dir/src/offline_node.cpp.o] Error 1
       > make[2]: *** [CMakeFiles/cartographer_ros.dir/build.make:135: CMakeFiles/cartographer_ros.dir/src/node.cpp.o] Error 1
       > make[1]: *** [CMakeFiles/Makefile2:174: CMakeFiles/cartographer_ros.dir/all] Error 2
       > make: *** [Makefile:146: all] Error 2
       For full logs, run 'nix log /nix/store/kvfyf4qdw4nvfb7fb35fnrkdpai0n2gw-ros-humble-cartographer-ros-2.0.9002-r1.drv'.
error: 1 dependencies of derivation '/nix/store/4i7igbb9d59b54ybnbcgmnfh660qqhy5-ros-env.drv' failed to build
error: 1 dependencies of derivation '/nix/store/cvc3yqb06qm28m6k2pagqvf145xx9lyn-ROS-2-Workspace-env.drv' failed to build
```

Using `develop` branch due to recently merged fixes for `cartographer`.

pjbuterbaugh, 2025-04-17:
`flake.nix`:

```
{
  inputs = {
    nix-ros-overlay.url = "github:lopsided98/nix-ros-overlay/develop";
    nixpkgs.follows = "nix-ros-overlay/nixpkgs";  # IMPORTANT!!!
  };
  outputs = { self, nix-ros-overlay, nixpkgs }:
    nix-ros-overlay.inputs.flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ nix-ros-overlay.overlays.default ];
        };
      in {
        devShells.default = pkgs.mkShell {
          name = "ROS 2 Workspace";
          packages = [
            pkgs.colcon
            pkgs.cairo

            # Main set
            pkgs.gazebo
            pkgs.python312
            pkgs.python312Packages.flake8 

            # ... other non-ROS packages
            (with pkgs.rosPackages.humble; buildEnv {
              paths = [
                ros-core
                # gazebo-model-attachment-plugin-msgs
                # gazebo-model-attachment-plugin
                # gazebo-ros-pkgs
                cartographer
                cartographer-ros
                # nav2-bringup

                # ... other ROS packages
              ];
            })
          ];
        };
      });
  nixConfig = {
    extra-substituters = [ "https://ros.cachix.org" ];
    extra-trusted-public-keys = [ "ros.cachix.org-1:dSyZxI8geDCJrwgvCOHDoAfOm5sV1wCPjBkKL+38Rvo=" ];
  };
}
```

wentasah, 2025-04-18:
Does  #622 fix the problems for you?

pjbuterbaugh, 2025-04-18:
Attempting to run this on my system, will report back.

pjbuterbaugh, 2025-04-19:
`cartographer-ros` builds and the shell launches. This fix works!

PR #622 (2025-04-27): cartographer-ros: Fix compile errors
Closes #621
closed 2025-04-27 by lopsided98