lopsided98/nix-ros-overlay #638: Last regeneration removed controller-manager, causes issue with rqt

Pleune, 2025-06-18:
See here - https://github.com/lopsided98/nix-ros-overlay/commit/2435baa8acbd4bf244551dff13a42978f8075319#r160318458

The `rqt-controller-manager` package depends on the non-rqt `controller-manager` giving this error line when evaluating:

```
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:34:12:
           33|
           34|   strict = derivationStrict drvAttrs;
             |            ^
           35|

       … while evaluating derivation 'nix-shell'
         whose name attribute is located at /nix/store/mbrmylpk0gv0kwdxibi5imfrgwlk5498-source/pkgs/stdenv/generic/make-derivation.nix:438:13

       … while evaluating attribute 'nativeBuildInputs' of derivation 'nix-shell'
         at /nix/store/mbrmylpk0gv0kwdxibi5imfrgwlk5498-source/pkgs/stdenv/generic/make-derivation.nix:490:13:
          489|             depsBuildBuild = elemAt (elemAt dependencies 0) 0;
          490|             nativeBuildInputs = elemAt (elemAt dependencies 0) 1;
             |             ^
          491|             depsBuildTarget = elemAt (elemAt dependencies 0) 2;

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: evaluation aborted with the following error message: 'lib.customisation.callPackageWith: Function called without required argument "controller-manager" at /nix/store/5v05g0yg77wl48nh9a7fd407k6wnswr0-source/distros/jazzy/rqt-controller-manager/default.nix:5'
```

hacker1024, 2025-07-07:
https://github.com/ros/rosdistro/pull/46633

liarokapisv, 2025-07-17:
@hacker1024 do you have more context on the controller-manager issue? The issue linked seems to be about python3-filelock

mbeutelspacher, 2025-07-17:
@liarokapisv Since controller-manager depends on python3-filelock (https://github.com/ros-controls/ros2_control/blob/master/controller_manager/package.xml#L36). My assumption would be that since this dependency could not be resolved, controller manager packages was removed from the overlay. Once the rosdistro MR is merged, the dependency can be resolved and controller manager will be part of this overlay again.

closed 2025-09-09 by wentasah