lopsided98/nix-ros-overlay #645: ardrone_autonomy

SCOTT-HAMILTON, 2025-07-09:
Hello, would it be possible to add https://github.com/AutonomyLab/ardrone_autonomy ?

From my understanding of https://wiki.ros.org/ardrone_autonomy, it's available in distros fuerte, groovy, hydro, indigo, jade and kinetic. None of which are included in this project.

Does that mean that it'll be impossible to add support for it ?

I'm happy to do it myself with a bit of guiding.

SCOTT-HAMILTON, 2025-07-10:
Hi, it's nothing important enough for me to feel comfortable about asking you to help beyond just a little guiding.

I'm just trying to interface with an old AR. Drone 2.0 from Parrot and if I can do so in a nix environment, it would be a must.

wentasah, 2025-07-10:
Hi,

I just tried whether the [ros2nix](https://github.com/wentasah/ros2nix/) tool can cope with this package. It turns out that some manual work will be needed. I run:

```
git clone https://github.com/AutonomyLab/ardrone_autonomy
ros2nix --distro=noetic $(find -name package.xml)
nix-build -A rosPackages.noetic.ardrone-autonomy
```

It turns out that the package needs  [ardronelib](https://github.com/AutonomyLab/ardronelib.git), which is packaged neither for ROS nor for Nix. So I modified the Nix expression generated by `ros2nix` (`ardrone_autonomy/package.nix`) to the following:

```nix
{ lib, buildRosPackage, SDL, binutils, camera-info-manager, catkin, daemontools, git, gtk2, image-transport, libxml2, message-generation, message-runtime, nav-msgs, roscpp, roslint, sensor-msgs, std-srvs, tf, udev, wirelesstools, which, ffmpeg }:
let
  ardrone-autonomy = buildRosPackage {
    pname = "ros-noetic-ardrone-autonomy";
    version = "1.4.1";

    src = ./.;

    buildType = "catkin";
    buildInputs = [ SDL binutils catkin daemontools git gtk2 libxml2 message-generation roslint udev wirelesstools ffmpeg ];
    propagatedBuildInputs = [ camera-info-manager image-transport message-runtime nav-msgs roscpp sensor-msgs std-srvs tf ];
    nativeBuildInputs = [ catkin which ];

    meta = {
      description = "ardrone_autonomy is a ROS driver for Parrot AR-Drone 1.0 and 2.0 quadrocopters. This driver is based on official AR-Drone SDK version 2.0.1.";
      license = with lib.licenses; [ bsdOriginal ];
    };
  };
in
lib.patchExternalProjectGit ardrone-autonomy {
  url = "https://github.com/AutonomyLab/ardronelib.git";
  originalUrl = "git://github.com/AutonomyLab/ardronelib.git";
  rev = "2f987029c55531e4c0119c3600f9c57f935851ed";
  fetchgitArgs.hash = "sha256-/EeiTs+6PjH8zdFrmBLYUr4156Voy3pDNIQPgz59lCo=";
}
```

With this, it "mostly" compiles, but fails with errors, which seem to be caused by ffmpeg incompatibility. I guess the package expects an older version of ffmpeg than what's in nixpkgs. Maybe, if you upgrade the code to be compatible with current ffmpeg, it will work.

SCOTT-HAMILTON, 2025-07-11:
Ok, got it working with a few patches (ffmpeg 4.x → ffpeg 7.x and more) [https://github.com/SCOTT-HAMILTON/ardronelib/commits?author=SCOTT-HAMILTON](https://github.com/SCOTT-HAMILTON/ardronelib/commits?author=SCOTT-HAMILTON).

How can I create an environment with this package included ?

SCOTT-HAMILTON, 2025-07-13:
Ok, I have another issue.

I didn't like `patchExternalProjectGit` because it made working on `ardronelib` unpractical:
To test in patch, I had to stage, commit, push, change rev, fakeSha, build, update sha.

It was too much steps.

So I got rid of it and made a separate flake that provides ardronelib's package.

To do so, I used this derivation:
```nix
{ lib
, stdenv
, pkg-config
, SDL
, gtk2
, libxml2
, udev
, wirelesstools
, ffmpeg
, nix-gitignore
}:

stdenv.mkDerivation {
  pname = "ardronelib";
  version = "13-08-2025";

  src = nix-gitignore.gitignoreSource [ ] ./.;

  nativeBuildInputs = [ pkg-config ];
  buildInputs = [ SDL gtk2 libxml2 udev wirelesstools ffmpeg ];

  makeFlags = [ "INSTALL_PREFIX=$(out)" ];

  postInstall = ''
    mkdir -p $out/include
    mkdir -p $out/lib

    # Copy headers
    find ARDroneLib -type f -name '*.h' | while read file; do
      dest="$out/include/$(dirname "$file")"
      mkdir -p "$dest"
      cp "$file" "$dest/"
    done

    # Copy libraries
    find "$out" -name "*.a" -exec mv {} "$out/lib" \;
  '';

  meta = with lib; {
    description = "Patched version of ARDroneLib 2.0.1 used in ardrone_autonomy project";
    homepage = "https://github.com/SCOTT-HAMILTON/ardronelib";
    maintainers = [ "Scott Hamilton <sgn.hamilton+nixpkgs@protonmail.com>" ];
    platforms = platforms.linux;
  };
}
```

And added this flake.nix:
```nix
{
  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
    flake-utils.url = "github:numtide/flake-utils";
  };
  outputs = {
    nixpkgs,
    flake-utils,
    ...
  }: 
[... 258 lines cut ...]
ros-noetic-ardrone-autonomy> make  -f CMakeFiles/ardrone_autonomy_generate_messages.dir/build.make CMakeFiles/ardrone_autonomy_generate_messages.dir/build
ros-noetic-ardrone-autonomy> make[2]: Entering directory '/build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/build'
ros-noetic-ardrone-autonomy> make[2]: Nothing to be done for 'CMakeFiles/ardrone_autonomy_generate_messages.dir/build'.
ros-noetic-ardrone-autonomy> make[2]: Leaving directory '/build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/build'
ros-noetic-ardrone-autonomy> [ 98%] Built target ardrone_autonomy_generate_messages
ros-noetic-ardrone-autonomy> In file included from /build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/include/ardrone_autonomy/ardrone_driver.h:41,
ros-noetic-ardrone-autonomy>                  from /build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/src/ardrone_driver.cpp:30:
ros-noetic-ardrone-autonomy> /build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/include/ardrone_autonomy/ardrone_sdk.h:51:10: fatal error: config.h: No such file or directory
ros-noetic-ardrone-autonomy>    51 | #include <config.h>
ros-noetic-ardrone-autonomy>       |          ^~~~~~~~~~
ros-noetic-ardrone-autonomy> compilation terminated.
ros-noetic-ardrone-autonomy> make[2]: *** [CMakeFiles/ardrone_driver.dir/build.make:79: CMakeFiles/ardrone_driver.dir/src/ardrone_driver.cpp.o] Error 1
ros-noetic-ardrone-autonomy> make[2]: Leaving directory '/build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/build'
ros-noetic-ardrone-autonomy> make[1]: *** [CMakeFiles/Makefile2:3736: CMakeFiles/ardrone_driver.dir/all] Error 2
ros-noetic-ardrone-autonomy> make[1]: Leaving directory '/build/0s62l4z8v9pcibvlhzp198ggk1sml726-source/build'
ros-noetic-ardrone-autonomy> make: *** [Makefile:136: all] Error 2
```

</details>

Which leads me to believe that it's the `generate_messages(DEPENDENCIES geometry_msgs std_msgs)` step of the CMakeLists.txt that doesn't succeed.

But why do I get this error now whereas I had no such issue when letting cmake build ardronelib itself.

How come the separation of the two builds produced such a behaviour ?

wentasah, 2025-07-13:
I don't think that the error is related to `generate_messages(DEPENDENCIES geometry_msgs std_msgs)`. Multiple build jobs probably run in parallel and the location of the error just happens to be  next to messages from another job.

If I understand what you're doing, `ardrone_sdk.h` tries to include `config.h`, which should be most likely at `ARDroneLib/Soft/Common/config.h`. I guess your include paths is just incorrect. Try adding this:
```nix
makeFlags = [ "VERBOSE=1" ];
enableParallelBuilding = false;
```
to `ardrone_autonomy/package.nix` to see the exact compiler flags for the failed command.

Another potential problem with your approach (which is probably unrelated to your error) is that you build `ardronelib` against different version of nixpkgs (current `nixos-unstable`) than the `ardrone_autonomy` package, which is built against an older version pinned in `nix-ros-overlay`. This can lead to various weird problems. I suggest not creating an independent flake for `ardronelib`, but building both packages from the same flake. Or, it might be easier to do this debugging without flakes at all, because flakes force you to add all files to git (is most cases). Another option, if you like flakes, is to use `nix build --impure` and refer to `ardronelib` via absolute path to its source code directory.

SCOTT-HAMILTON, 2025-07-14:
Ok, I feel stupid for being fooled by such an elementary trick: parallel building logs, and also for the simplicity of the solution.

Turns out I just didn't write the ardronelib package correctly,

replacing
```nix
        packages.default = pkgs.rosPackages.noetic.callPackage ./package.nix {
          inherit ardronelib;
        };
 ```
By
```nix
        packages.default = pkgs.rosPackages.noetic.callPackage ./package.nix {
          ardronelib = ardronelib.packages.${system}.default;
        };
```
solved it...