lopsided98/nix-ros-overlay #649: nav2-costmap-2d does not build on jazzy

jonboh, 2025-09-06:
It seems to be due to a warning on gcc that now is an error. With this patch I get it to build:
```
nav2-costmap-2d = jazzy-prev.nav2-costmap-2d.overrideAttrs (super: {
  postPatch =
    (super.postPatch or "")
    # it seems that this was previously a warning and now a hard error on gcc
    + ''
      # use of layers_ = {0};
      substituteInPlace include/nav2_costmap_2d/denoise/image_processing.hpp \
        --replace-fail \
          "labels_ = {0};" \
          "labels_.clear();
           labels_.push_back(0);"
    '';
```

wentasah, 2025-09-07:
Thanks for the report. `nav2-costmap-2d` builds fine in my fork. See #651.

jonboh, 2025-09-07:
Actually, checking back again, this issue only appears in the latest (navigation2) syncs with superflore, not on `nix-ros-overlay/master`, my bad.
This fix might be needed in the next sync.

wentasah, 2025-09-09:
This repo has the latest ROS sync now and `jazzy.nav2-costmap-2d` builds. Please repoen, if there are still some problems regarding nav2-costmap-2d.

closed 2025-09-09 by wentasah