lopsided98/nix-ros-overlay #650: nav2-rviz-plugins fails due to missing dependency on startup due to nix hardening

jonboh, 2025-09-06:
Allowing lazy dynamic loading fixes the error, the panel is usable.
```
nav2-rviz-plugins = jazzy-prev.nav2-costmap-2d.overrideAttrs (super: {
  hardeningDisable = ["bindnow"]; # allow lazy loading dependencies, otherwise breaks with an undefined symbol: _ZN17nav2_rviz_plugins12DockingPanel9onStartupEv
});
```

wentasah, 2025-09-07:
Ahh, interesting. You've found the reason behind this problem, which appeared in several packages! We tried to [fix it upstream by removing unused Qt slots](https://github.com/ros-navigation/navigation2/pull/4974). Which branch/version are you using? I mean, did they introduce other unused slots are you are using a version before the mentioned PR?

jonboh, 2025-09-07:
Your fix ([e6f500](https://github.com/ros-navigation/navigation2/commit/e6f500e5b7528737f4a883598293b62c72c83946)), seems to be present only on `main`, `humble_main` and `kilted`.