lopsided98/nix-ros-overlay #651: Maintenance status?

wentasah, 2025-09-07:
Hi @lopsided98,

it's almost four months since you last activity in this repo. The current version becomes more and more stale and does not even contain the latest ROS release Kilted. There are unmerged PRs for many of these problems. I hope that are all right and that this is only a temporary situation (it seems you are still active in nixpkgs).

It would be great if you can comment on the situation and your future plans about this repo. Maybe it's time for setting up a Github organization with a maintenance team, where more than one person would be able to merge PRs. I'm definitely willing to join and help.

Since I'm not sure I'll get any response, I've already **created a fork**: https://github.com/wentasah/nix-ros-overlay, where I merged some of the PRs and synced it with the latest ROS distro releases. The current status can be observed in [my hydra](https://hydra.iid.ciirc.cvut.cz/project/nix-ros-overlay-wentasah). I'll continue maintaining the fork and I plan to merge PRs and address issues from here. If we won't here from you for other few months, I'll probably move my fork to a new github org and announce it as "official". However, I'd prefer to continue using this repo (perhaps after moving to an org) as the official one.

lopsided98, 2025-09-07:
I'm fine, I just don't use ROS anymore and therefore this project has become a low priority. I've always been reluctant to add additional committers mostly due to the trust involved. You are someone who has invested a lot of time into this and made good contributions for over two years, so I think that establishes a reasonable level of trust. If I were to assume malicious intent, it would have to be a long and dedicated con 😃

Therefore, I will add you as a committer. I'd also be open to moving this repo to an organization and potentially adding other committers, but this should at least allow things to get fixed for now.

wentasah, 2025-09-07:
Great! I didn't expect so quick answer :-) Thanks.

In the upcoming days, I'll go over the PRs and merge them here so that this repo becomes up to date. I'll try to follow your maintenance style as I learned that it has many advantages. For more involved changes and problems, I'll use PRs and issues as before so that you or anybody else has a chance to comment. If you're not happy about anything I'll do, just let me know and I'll try to change that.

closed 2025-09-07 by wentasah